steveicarus/iverilog #364: Ctrl-D causes unresponsive interactive vvp session

ercanal, 2020-08-15:
If Ctrl-D is pressed in an interactive vvp session( Ctrl-C is pressed and prompt is shown), It is not possible to get prompt with another Ctrl-C. It prints same message with different time tick values.
```
^C** VVP Stop(0) **
** Flushing output streams.
** Current simulation time is 2175274758 ticks.
> ** Continue **
````

martinwhitaker, 2020-08-15:
Ctrl-D acts like the cont command, allowing simulation to continue. Try this simple test program:
````
module test;

integer count = 0;

always begin
  #1 count = count + 1;
  $display(count);
end

endmodule
````
Ctrl-C will interrupt the count output and show the interactive prompt. Ctrl-D will resume the count output.

ercanal, 2020-08-16:
I tested your test case, it works fine in Windows,   but in WSL2 session after first Ctrl-D I couldn't get prompt back again.

martinwhitaker, 2020-08-16:
Works here in two different Linux distributions (Mageia 7 and Ubuntu 18.04). So it will need someone who uses WSL2 to investigate why it's not working there.

ercanal, 2020-08-28:
Related bug for WSL  https://github.com/microsoft/WSL/issues/3421

caryr, 2021-01-20:
Closing this as not an Icarus bug.

closed 2021-01-20 by caryr