steveicarus/iverilog #366: if macro used as a parameter in another macro

ercanal, 2020-08-21:
Unexpected output. from: [https://stackoverflow.com/questions/15373113/how-to-create-a-string-from-a-pre-processor-macro](https://stackoverflow.com/questions/15373113/how-to-create-a-string-from-a-pre-processor-macro)

output is `HPATH instead of top.chip.block

```systemverilog
`define STRINGIFY(x) `"x`"
`define HPATH top.chip.block
module tb();
    string hpath;
    initial begin
    hpath = `STRINGIFY(`HPATH);
    $display(hpath);  
    end
endmodule
```

ercanal, 2020-08-21:
More interesting for me is that following code is working because there is no double quotation in define lines.

```systemverilog
`define STRINGIFY(x) x
`define HPATH top.chip.block
module tb();
    string hpath;
    initial begin
    hpath = ``STRINGIFY(`HPATH);
    $display(hpath);  
    end
endmodule
```

martinwhitaker, 2020-08-24:
Fixed in the master branch.

closed 2020-08-24 by martinwhitaker