steveicarus/iverilog #369: Does Icarus support rand?

mickey4u, 2020-09-09:
I am using `rand` in my system verilog testbench and I am getting an error. Does Icarus support `rand`? if not what will be the alternative in achieving this?

martinwhitaker, 2020-09-10:
No, it doesn't. You can declare random variables, but there is no support for the randomize() method. You will have to implement your own randomize() method, using the standard probabilistic distribution system functions (e.g. $random) to provide the random values.

mickey4u, 2020-09-10:
@martinwhitaker Thanks for this. How do you declare random variables that Icarus supports.

martinwhitaker, 2020-09-10:
As per the SV standard - prefix a class property declaration with the keyword rand. But of course, without support for randomize(), that is just decoration.

mickey4u, 2020-09-10:
@martinwhitaker Okay I had that but Icarus flagged an error.
`    typedef struct packed {
        rand bit [15:0] a;
        bit [15:0] b;
    } cc;`

`
testbench.sv:42: syntax error
testbench.sv:42: Error in struct/union member.
`

svenka3, 2020-09-11:
@mickey4u rand is a valid qualifier only inside class and not struct