steveicarus/iverilog #372: Segmentation fault if -I and -M are used together

kentindell, 2020-09-25:
Very easy to reproduce:

```
$ iverilog -I.. -M..
Segmentation fault (core dumped)
```

Also applies to `-Mall`.

martinwhitaker, 2020-09-25:
Are you using the master branch? I get
````
% iverilog -I.. -M.. 
iverilog: can't open .. file.
````
I do get a segfault with the v10 branch, but that was fixed in the master branch by pull request #154.

kentindell, 2020-09-25:
OK, I've just re-cloned and built everything from scratch. The -I/-M now doesn't segfault, but `-Mall=..` does.

```
$ iverilog -I.. -Mall=..
Segmentation fault (core dumped)
```

kentindell, 2020-09-25:
By the way, I only discovered this because I'm using cocotb, and one of the problems with paths is actually down to the cocotb build system requiring a make clean between different tests harnesses. I learned this the hard way, so if anyone sees this with a similar problem, the fix is: `make clean` :)

martinwhitaker, 2020-09-25:
I still can't reproduce that, with a clean build from the head of the master branch:
````
% iverilog -I.. -Mall=..
iverilog: can't open .. file.
````
Please run
````
gdb -ex run --args iverilog -I.. -Mall=..
````
and at the (gdb) prompt after the segfault, use the `bt` command to get the stack trace.

kentindell, 2020-09-25:
OK, there's something really weird going on. It seems `which` is broken, since it reports `/usr/local/bin/iverilog` but when I use `-v` it's clearly not running it.. I think this is an Ubuntu snafu. I've run \iverilog` with an explicit path directly and can confirm that v11 doesn't segfault. I'll close this issue.

closed 2020-09-25 by kentindell