steveicarus/iverilog #376: Enable waveform dumping via command line

alexforencich, 2020-10-16:
As a recent user of cocotb, the lack of ability to command iverilog to create a dump file with command line arguments has become a rather serious problem.  When simulating, cocotb does not create a top-level verilog file, it simulates the top-level module directly.  The problem is that in this case, enabling waveform dump requires adding code to the actual module, not to the module's testbench.  The cocotb documentation recommends making this selectable based on a define.  The problem is that this does not work if the module contains other modules which have the same dump code added based on the same define.  

The most reasonable fix is to add a --dumpfile parameter or similar that can not only specify the dump file, but also enable dumping for the top-level module.

alexforencich, 2020-10-16:
Verilator accepts --trace during compilation, which results in creating a file dump.vcd during simulation.  It seems like it should be relatively straightforward to do something similar in icarus verilog, perhaps -trace in plusargs could trigger the equivalent of $dumpfile(top_module_name); $dumpvars(top_module_name, 0); .

martinwhitaker, 2020-10-16:
Why not add an extra module to your simulation that just calls $dumpfile and $dumpvars?

alexforencich, 2020-10-16:
I tried that already, but it didn't generate a dump file.  Not sure why.  I will try a few more iterations and see if I can figure out what I might have missed.

alexforencich, 2020-10-16:
So, I think the problem is that there is no way that I am aware of to add an additional module to the simulation.  I created an additional verilog file with a bare module and an initial block that calls $dumpfile and $dumpvars, but as nothing instantiates the module the dump file is not created.

martinwhitaker, 2020-10-16:
cocotb automatically adds `-s $TOPLEVEL` to the iverilog command line, so you need to add another `-s` flag to include another top-level module. Assuming a module named dump, in your Makefile, add
````
VERILOG_SOURCES += <path to>/dump.v
COMPILE_ARGS += -s dump
````

alexforencich, 2020-10-16:
Aha, that worked.  Didn't know it was possible to have multiple top-level verilog files.  Now I just need to set up the makefile to write out the extra verilog file with the correct module names.

alexforencich, 2020-10-16:
Template makefile to enable icarus verilog waveform dumps for cocotb:

    TOPLEVEL_LANG = verilog

    SIM ?= icarus
    WAVES ?= 0

    COCOTB_HDL_TIMEUNIT = 1ns
    COCOTB_HDL_TIMEPRECISION = 1ps

    TOPLEVEL = module
    MODULE   = test_$(TOPLEVEL)
    VERILOG_SOURCES  = ../../rtl/$(TOPLEVEL).v

    # module parameters
    SOME_PARAM=1

    ifeq ($(SIM), icarus)
        PLUSARGS += -lxt2

        COMPILE_ARGS += -P $(TOPLEVEL).SOME_PARAM=$(SOME_PARAM)

        ifeq ($(WAVES), 1)
            VERILOG_SOURCES += iverilog_dump.v
            COMPILE_ARGS += -s iverilog_dump
        endif
    else ifeq ($(SIM), verilator)
        COMPILE_ARGS += -Wno-SELRANGE -Wno-WIDTH

        COMPILE_ARGS += -GSOME_PARAM=$(SOME_PARAM)

        ifeq ($(WAVES), 1)
            COMPILE_ARGS += --trace
        endif
    endif

    iverilog_dump.v:
        echo 'module iverilog_dump();' > $@
        echo 'initial begin' >> $@
        echo '    $$dumpfile("$(TOPLEVEL).lxt");' >> $@
        echo '    $$dumpvars(0, $(TOPLEVEL));' >> $@
        echo 'end' >> $@
        echo 'endmodule' >> $@

    include $(shell cocotb-config --makefiles)/Makefile.sim

r2com, 2023-04-23:
sorry but it does not work for me for some reason. 
so I do have DUT Verilog file top.v as a top module named "dut_top" then i created another file dut_dump.v which has:

```
module dump (

);

initial
 begin
    $dumpfile("test.vcd");
    $dumpvars(0);
 end
endmodule
```

After that I compile it:
```
iverilog -o dut_top.vvp -s dut_top -s dump ./src/top.v ./src/dut_dump.v
vvp -M ../vip/build/ -m ../vip/build/libdut_top.vpi.so ./dut_top.vvp
```

The test.vcd file is produced, and it even has the relevant signals of top module, however; for some reason start and end time is 0 sec. 

So what is wrong here?

(btw its similar test design and the VPI C program is same and works and produces the vcd file with similarly written VHDL file with GHDL, so in other words, the VPI C program is correct)