steveicarus/iverilog #378: Can icarus save simulation state to a file and resume later?

mkhazraee, 2020-10-20:
Hi,

I'm simulating some open IP and it takes 10us of simulation time (or about 10 minutes of runtime) to get setup. Unfortunately there is no parameter to reduce the setup time, and each time I make a modification to the testbench I have to rerun the simulation from the beginning. I was wondering if there is a way to run the simulation once until setup time, save it to a file, and then for each testbench read the state from that file and continue the simulation. I'm making the testbench in python and cocotb, so the verilog part doesn't change between the runs.

Thanks,
Moein

themperek, 2020-10-20:
Not sure if this is possible with icarus but it is possible with cocotb. Example here: https://github.com/themperek/apbi2c_cocotb_example/blob/parallel/tb/checkpoint.py This was working at some point but needs adjustment for the current version (can be also more user friendly).

steveicarus, 2020-10-20:
This is a good question for the iverilog-devel mailing list, but no, Icarus
Verilog doesn't support checkpoints.

mkhazraee, 2020-10-20:
Thanks a lot @themperek, that's exactly what I was looking for. I figured the required updates to be compatible with current version, but still I have to make a better implementation because for large designs it gets "maximum recursion depth" error. I'll make it more user friends and would possibly make a pull request to add it to cocotb for others to use. 

@steveicarus I see, thanks. For now the cocotb solution totally works for me and it might provide more flexibility, but agreed that it would be really cool to have it in Icarus in the future.

closed 2020-10-20 by mkhazraee