steveicarus/iverilog #379: Licensing LGPL or GPL

neeldug, 2020-10-22:
I'm unsure of what the licensing of Icarus Verilog is, as the LICENSE file headers indicate GPL, whereas the `COPYING.lesser` file indicates the software is LGPL.

martinwhitaker, 2020-10-22:
The licence notice in each file indicates the license for that file. Most of the,source files are licensed under the GPL v2 (or later), the full text of which is found in the COPYING file.

The vlog95 target outputs some generated Verilog code in addition to the translated input code. That generated code is licensed under the LGPL v2.1, the full text of which is found in the COPYING.lesser file.

It's not helpful that GitHub shows the project is licensed under the LGPL. I don't know if @steveicarus can fix that - I can't.

steveicarus, 2020-10-22:
I do not know why it says LGPL, and then shows the GPL license COPYING
file. It's a mystery to me, and I can't seem to fix it.

martinwhitaker, 2020-10-23:
GitHub uses the `licensee` Ruby gem to detect the license. Trying that locally, I find it detects both the COPYING and COPYING.lesser licenses, but reports the COPYING.lesser license first. We could fix this by moving the COPYING.lesser file into the tgt-vlog directory (which is the only place the LGPL is used).

caryr, 2020-11-15:
I moved the COPYING.lesser file to the tgt-vlog95 directory so we will see if that fixes github.

martinwhitaker, 2020-11-15:
The licensee gem now reports the correct license:
````
% licensee --remote steveicarus/iverilog
License:        GPL-2.0
Matched files:  COPYING
COPYING:
  Content hash:  3becd209e8ed8039656c1debe01dd17b9a79208f
  Confidence:    100.00%
  Matcher:       Licensee::Matchers::Exact
  License:       GPL-2.0
````
but the project page is still showing the old result,

caryr, 2020-11-15:
I hope it's just that the cached value only gets updated every once in a while. I suppose we could argue the gem should report the most pessimistic instead of the most liberal license, but I can see it for either order. If things do not get updated before too long maybe we need to contact a github admin.

steveicarus, 2020-11-27:
The github page is still showing LGPL. Is there any tech support at Github that we can contact?

caryr, 2020-11-27:
I was looking at this the other day and I think there is a forum section we may be able to post this in. It's behavior is completely broken since it reports LGPL, but points to the GPL file.
Cary

caryr, 2020-11-29:
Here is the link to the forum: https://github.community/. I assume since this is owned by Steve his request may have more weight for this issue. I also assume you can just point to this report for the details.

steveicarus, 2020-11-30:
Here is the issue I created.

https://github.community/t/github-reporting-incorrect-license-on-home-page/146104

umarcor, 2020-11-30:
It might be faster to [contact](https://support.github.com/contact) directly, and ask them to please update the license detection for this repo. Some time ago, that was required for GHDL too.

caryr, 2020-12-07:
Yes, it looks like the community is a no go for resolving this issue. Hopefully Steve can contact them directly to get resolution.

caryr, 2020-12-28:
Steve are you going to contact them directly about this?

steveicarus, 2020-12-28:
> Assigned #379 <https://github.com/steveicarus/iverilog/issues/379> to

I guess that's my job.
Yeah, I'll contact them.

steveicarus, 2020-12-30:
Contacted Github.

Apparently, the license is detected once then cached. When we fixed the COPYING file, it never rescanned the License. Github support was able to force a rescan, and now we're showing the proper license.

closed 2020-12-30 by steveicarus