steveicarus/iverilog #380: Bugs in VPI interface

mkhazraee, 2020-10-22:
I'm trying to build the checkpoint feature for cocotb+Icarus (#378 ). During this process I ran into some issues and after talking with cocotb folks it seems to be Icarus bugs in VPI. I've made MRFE for the 3 cases based on simple_dff example of cocotb. 
[Icarus_VPI_MRFE.zip](https://github.com/steveicarus/iverilog/files/5424274/Icarus_VPI_MRFE.zip)

There are 3 cases:

1) If the instant name has the same name as a generate block inside that module (inside the module that is instantiated), this causes an infinite loop in the search process. You can reproduce the bug by using other_dff.sv in the MRFE. 

To be more specific, @ktbarret made a more simplified test and then @garmin-mjames suggested a potential cause for the bug:
"Looks like for some reason icarus is adding the this_name vpiModule handle 4 times instead of the 4 vpiScope handles this_name[0] through this_name[3].

I added a call to vpi_get_str(vpiFullName, new_hdl) in create_gpi_obj_from_handle and it's returning dff.this_name for each one no matter how deep it goes:
```
VpiCbHdl.cpp:893  in next_handle                     vpi_scan found 'dff.this_name.this_name.this_name.this_name'
VpiImpl.cpp:266  in create_gpi_obj_from_handle      VPI: Created GPI object dff.this_name from type vpiModule(32)
```
Icarus doesn't have a proper gen scope array scope, only gen scope. I don't know if this is related to the apparent bug."

(please refer to https://github.com/cocotb/cocotb/issues/2150 for more details).

2) If there is an always block that has a name, e.g. always @(posedge clk) begin: some_name, this results in a warning that it's unable to create some_name via any registered implementation. You can reproduce the bug by using other_dff2.sv in the MRFE. 

For this one @garmin-mjames mentioned "Icarus is returning vpiNamedBegin rather than vpiAlways (which is supported)"

(please refer to https://github.com/cocotb/cocotb/issues/2152 for more details).

3) If there is a function inside a module, I get VPI: Implementation does not know how to iterate over vpiFunction(20). You can reproduce the bug by using other_dff.sv in the MRFE. I found that it recognizes function as a MODULE which I think it shouldn't. 

Also @ktbarret mentioned: "we shouldn't be ever finding a function. Where it fails we call vpi_get(vpiType, handle). Where handle was previously discovered as not a vpiFunction. This implies that running that on the same handle returns a different result. This may be another bug."

I think it would be very beneficial if we can fix these issues on the VPI interface, and be able to revive the checkpoint feature for cocotb as an extension. For my own case, this feature is the most beneficial for Icarus, as it's slower compared to the QuestaSim or other not-open-source tools.

martinwhitaker, 2020-10-22:
Sorry, but to investigate these bugs, I'd want some stand-alone test cases, not something that requires cocotb.

mkhazraee, 2020-10-22:
Can you give me an example of an standalone case for this case? Maybe I'm missing something, but isn't VPI the interface between Verilog simulator and C? If you want to make something based on C instead of python I can look into it, potentially using cocotb's C files, but that's still not stand-alone as all.

martinwhitaker, 2020-10-22:
Standalone as in only uses Verilog and C/C++, not some Python framework. See https://iverilog.fandom.com/wiki/Using_VPI for a simple example. More examples can be found in https://github.com/steveicarus/ivtest/tree/master/vpi

mkhazraee, 2020-10-23:
I found that the scope test is very close to what I'm looking for. So I made a test based on that, but seems m1 or 2 problems are actually from cocotb. Here is the test and its output:
[vpi.zip](https://github.com/steveicarus/iverilog/files/5427542/vpi.zip)

I added my D-FF code and added some cases to their verilog. The only change to C is to print fullname instead of name, and added a case for vpiGenScope. 

For always block name it's indeed vpiNamedBegin rather than vpiAlways. If vpiNamedBegin is what is expected, how can we figure out it's an always block? is there another pointer or method to check that?

I still don't know how vpiFunction is returned as vpiModule, but I guess that's something on the cocotb side.

martinwhitaker, 2020-10-23:
When iterating over internal scopes, it is correct to return vpiNamedBlock. It is the named block that introduces the new scope, not the always construct. And this is confirmed by the object data model diagrams in the LRM (e.g. IEEE 1364-2005 section 26.6.3).

With a full VPI implementation, it would be possible to determine whether the named block belonged to an always construct by getting a handle to the associated process and querying its object type, but Icarus Verilog only implements a subset of the VPI, and that functionality is not included.

ktbarrett, 2020-10-23:
@martinwhitaker If returning `vpiNamedBegin` is more appropriate in this case, then it is cocotb that needs to support this behavior. We don't really need to know it's an always block, we just expected `vpi_get_handle_by_name` to return `vpiAlways` on a named always block.

We have determined that issue 1 is a logic bug in cocotb. Issue 3 still needs to be tracked down on our end.

mkhazraee, 2020-10-23:
Thanks @martinwhitaker for your prompt responses. I'm gonna close the issue, as it seems to be on the cocotb side.

closed 2020-10-23 by mkhazraee