steveicarus/iverilog #381: Icarus performance with bitwise operators and binary operators.

mickey4u, 2020-10-22:
In terms of performance if you have say two inputs a and b with the same bit width say 64. If you apply a bitwise operator and binary arithmetic operator, it's takes the parity circuit longer simulation time compared with that of the adder.
For example, I have an adder like this `{c_o, sum} = a + b + c` where `a` and `b` 64 bit and a parity circuit say `parity = parity ^ data[i]` where data = 128 bit. Testing these two circuits using icarus I observed the parity circuit takes longer. Any reason why it will take parity circuit longer. Is Icarus optimized for certain operators than others?

martinwhitaker, 2020-10-22:
Your two code fragments look completely different. Try providing complete examples to show what you are really doing.

mickey4u, 2020-10-22:
@martinwhitaker Sorry didn't make the question clear enough. What I actually meant was that when I change the parameter and apply my testvectors to the adder and parity circuits, I observe no much of a difference in terms of simulation time for the adder circuit. But in the case of the parity circuit looks as if the size of the inputs has an effect on the simulation time. So my question is if there is any reason for that? Are certain operators optimized compared to others in Icarus Verilog?
```
module adder #(parameter width=64)(input logic [width-1 : 0] a, b, input  cin, output logic cout, output logic [width-1:0] sum); 
 always @(a, b or carry_in) begin 
 {cout, sum} = a + b + cin;
 end 
 endmodule

module eparity #(parameter width=128)(input logic [width-1 : 0] data, output logic p); 
 logic [width : 0] i_parity; 
 genvar i; 
 assign intmd_parity[0] = 0; 
 for(i = 0; i < width; i = i + 1) begin 
 assign i_parity[i+1] = data[i] ^ i_parity[i]; 
 end 
 
 assign p = i_parity[width]; 
 endmodule

```

martinwhitaker, 2020-10-23:
Why do you find this surprising? In the first case you are doing a single multi-bit add. In the second case you are generating `width` single-bit continuous assignments. In the first case the simulator has to process one evaluation event each time one of the inputs changes. In the second case the simulator has to process `2 * width` evaluation events each time the input changes.

If you want to write more efficient code, you should learn about Verilog reduction operators.

mickey4u, 2020-11-14:
@martinwhitaker Thanks. But why not `width` evaluation events but instead `2*width` evaluation events (I'm aware this is not the right place for such questions though).

caryr, 2020-11-15:
Because each XOR is sensitive to both the previous parity and the data bit being checked. The eparity routine is horrible from a memory and execution standpoint. Like Martin said the reduction operator is the correct way to implement this which is a single assignment.

`assign p = ^data;`

This is not a bug in Icarus, so I am closing this.

closed 2020-11-15 by caryr

martinwhitaker, 2020-11-15:
In fact, depending on the order in which the simulator evaluates events and on how the individual bits change, it could require anywhere between `width` and `width*(width - 1)` evaluation events, as each bit change potentially ripples up to the MSB.