steveicarus/iverilog #383: Dynamic array assignment should copy the array, not pass a reference.

martinwhitaker, 2020-10-27:
Consider the following example:
````
module test;
  logic [7:0] i, x[], y[];

  initial begin
    x = new [4];
    for (i = 0; i < 4; i = i + 1) x[i] = 1 + i;
    y = x;
    for (i = 0; i < 4; i = i + 1) y[i] = 4 - i;

    $display(x[0],,x[1],,x[2],,x[3]);
    $display(y[0],,y[1],,y[2],,y[3]);
  end

endmodule

````
The IEEE standard clearly states that `y` should be a copy of `x`, and hence the expected output is
````
  4   3   2   1
  1   2   3   4
````
whereas we get
````
  4   3   2   1
  4   3   2   1
````
Other simulators produce the expected result.

The same holds true when passing arguments to subroutines.

steveicarus, 2020-11-23:
I'll give this a try.

steveicarus, 2020-11-26:
Fixed in pr #396

closed 2020-11-26 by steveicarus