steveicarus/iverilog #384: Asynchronous reset generated using FPGA logic reset trick...

wm2015email, 2020-11-02:
Let's suppose I have a Flip-Flop with a gated clock....  I'm using the code below in an FPGA to reset my flip-flops... (this works in an FPGA synthesis compile because Flip-flops default to zero state when locating the FPGA image file...)  However, when I simulate the correct way using `define MODELSIM , then icarus verilog doesn't reset my flip-flops but modelsim does....

```
`timescale 1ns/10ps
module zynq_pl_genrst(
    input  wire sys_clk,
    output wire sys_rstn
);

localparam W = 40;

// FPGA SYNTHESIS TRICK - DON't USE IN AN ASIC...
`ifdef MODELSIM
    reg [W-1:0] d = {1'bX, {W-1{1'b0}}};
`endif

`ifdef IVERILOG
    reg [W-1:0] d = {1'b0, {W-1{1'b0}}};
`endif

always @(posedge sys_clk) begin
    d[0]     <= 1'b1;
	d[W-1:1] <= d[W-2:0];
end

assign sys_rstn = d[W-1];

endmodule

```

example:

```
// Sorry, i don't have to test this code... but this is basically is my attempt to recreate what is happening in my simulation...
// and to show difference between icarus and modelsim...
module test;

reg clk;
wire rstn;

initial begin
    clk = 1'b0;
   forever begin
      #10 clk = 1'b0;
      #10 clk = 1'b1;
   end
end

zynq_pl_genrst genrst(
   sys_clk  (clk),
   sys_rstn (rstn)
);

dff dff1(
    input wire clk,
   input wire arstn,
   input wire [7:0] d,
  output wire [7:0] q
);
   always @(posedge clk or negedge arstn) begin
       if (!rstn) begin
           d <= 8'd0;           
       end
       else begin
           d <= q;
       end
   end
endmodule

reg          arstn;
wire [7:0] din = 8'hA;
wire [7:0] dout;

// how its used
dff dff1(
    .clk  (1'b0),  //gated clock
    .arstn (arstn), //intenrally generate FPGA reset that goes high x cycles after FPGA configuration
    .d  (din),
   .q   (dout)  //dout should reset in icarus... when `modelsim is defined
);

endmodule

```

martinwhitaker, 2020-11-02:
After removing the flip-flop instantiation (which you don't provide code for) and fixing the syntax of the genrst instantiation, with IVERILOG defined I see rstn starts at 1'bx, goes to 1'b0 at time 20, and goes to 1'b1 at time 800. With MODELSIM defined rstn starts at 1'b0 and goes to 1'b1 at time 800. So there is no problem there. I think your issue is in the implementation of dff, most likely a time-0 race.

martinwhitaker, 2020-11-24:
Without seeing the code for the flip-flop, we can't help you further with this, so closing. Feel free to reopen if you can provide more information.

closed 2020-11-24 by martinwhitaker