steveicarus/iverilog #385: always @* for constant assignments

bfg86, 2020-11-03:
Using always @* for constant assignments is giving me trouble:

```
module test ();
  reg y;
  always @* begin
    y = 1'b0;
  end
  initial begin
    #1;
    $display("Value of y: %b", y);
  end
endmodule
```
```
iverilog -Wall '-g2012' test.v
test.v:3: warning: @* found no sensitivities so it will never trigger.
./a.out
Value of y: x
```
I would expect it to evaluate at time=0

svenka3, 2020-11-03:
I believe Icarus is correct as per LRM here. The behavior you are looking for is available with SV always_comb, see below code:

```

module test ();
  reg y;
  reg y_sv;

  always @* begin
    y = 1'b0;
  end

  always_comb begin
    y_sv = 1'b0;
  end
  initial begin
    #1;
    $display("Value of y: %b", y);
    $display("Value of y_sv: %b", y_sv);
  end
endmodule

```

Icarus gives:

```
iverilog -g2012 ../vlog_src/alw_star.v 
Value of y: x
Value of y_sv: 0

```
HTH
Srini

bfg86, 2020-11-03:
The code I'm running is generated by sv2v from a SV construct like this: (sorry haven't tested this code)
```
logic y;
generate if (FF)
  always @(posedge clk) y <= some_input;
else
  assign y = 1'b0;
endgenerate
```
Anyways I can post-process the sv2v output to change always @* to always_comb. I think the other tools I use will allow this (yosys, vivado and an old version of quartus).

svenka3, 2020-11-03:
You may want to try this workaround to stay with Verilog should you have to:

https://sourceforge.net/p/iverilog/bugs/864/

martinwhitaker, 2020-11-03:
@svenka3 is correct.

closed 2020-11-03 by martinwhitaker