steveicarus/iverilog #386: Erroneous explicit cast required error

jaminel, 2020-11-18:
The following code results in an erroneous "error: This assignment requires an explicit cast."
This is using latest main branch.

Thanks,
-Rad

-------------- test.sv ----------------
module test;
   typedef enum logic [8:0]
     {ILLEGAL,
      IA,
      IB
      } inst_t;
   inst_t        ipb_inst;
   
   typedef struct packed
     {//
      inst_t        inst;
      logic 	    iw;
     } ipb_data_t;
   ipb_data_t ipb_d;

   always
     ipb_inst = ipb_d.inst;

endmodule

martinwhitaker, 2020-11-18:
The compiler converts an access to a packed struct member into a part select of the entire struct, losing the member type information on the way. This will probably also lead to obscure bugs when a packed struct contains a mixture of 2-state and 4-state members.

jaminel, 2020-11-19:
Everything in this case is 4-state.  Note that changing to an assign command does not have the issue, but we can't implement a flip-flop with an assign.  Perhaps knowing that could help find solution.
< always
<   inb_inst = inb_d.inst
---
> assign inb_inst = inb_d.inst

martinwhitaker, 2020-11-19:
The comment about mixed 2- and 4- state was a reminder to whoever fixes this bug to consider the wider ramifications.

If a continuous assignment doesn't have this issue, that probably means that the compiler is not performing the necessary assignment compatibility checks in that case.

martinwhitaker, 2020-11-20:
Fixed in both the master and v11 branches. Thanks for reporting this.

closed 2020-11-20 by martinwhitaker