steveicarus/iverilog #387: SV: super.new() not compiling

svenka3, 2020-11-22:
Using 

>> Icarus Verilog version 12.0 (devel) (s20150603-961-g292d174)

Is simple, basic OOP supported? I get the below error for super.new() (Code below). Any clues?

Thanks
Srini

Error:
../ivl_uvm_test_class.sv:13: assert: PFunction.cc:47: failed assertion statement_

```verilog

package test_pkg;

  virtual class uvm_void;
  endclass : uvm_void

  class uvm_object extends uvm_void;
    virtual function void print ();
      $display ("Print");
    endfunction : print 
  endclass : uvm_object

  class uvm_report_object extends uvm_object;
    function new ();
      super.new ();
    endfunction : new 
  endclass : uvm_report_object 

endpackage : test_pkg

module m;
  import test_pkg::*;
  uvm_object u0;
  
   initial begin : test
     #100;
     $display ("Hello World");
     u0 = new ();
     u0.print();

   end : test

endmodule : m

```

martinwhitaker, 2020-11-22:
Simple OOP is supported, because it is used in the test suite, e.g.

https://github.com/steveicarus/ivtest/blob/master/ivltests/sv_class21.v

The problem with your code is that you call super.new(), but the base class does not explicitly define a new() function. That's a bug in the compiler.

svenka3, 2020-11-22:
Thanks for the quick reply. I have a variant as below - even with new() in all classes, it fails. Should I open a separate issue?

Regards
Srini

```verilog
package test_pkg;

  virtual class uvm_void;
    function new ();
    endfunction : new 
  endclass : uvm_void

  class uvm_object extends uvm_void;
    function new ();
    endfunction : new 

    virtual function void print ();
      $display ("Print");
    endfunction : print 
  endclass : uvm_object

  class uvm_report_object extends uvm_object;
    function new ();
      super.new ();
    endfunction : new 
  endclass : uvm_report_object 

endpackage : test_pkg

module m;
  import test_pkg::*;
  uvm_object u0;
  
   initial begin : test
     #100;
     $display ("Hello World");
     u0 = new ();
     u0.print();

   end : test

endmodule : m

```

steveicarus, 2020-11-23:
Fixed here:
https://github.com/steveicarus/iverilog/pull/394

closed 2020-11-23 by steveicarus