steveicarus/iverilog #390: SV: Using string argument with default value gives error

svenka3, 2020-11-22:
IVL 12.0 devel. Using string argument with default value gives error:

> ../ivl_i390.sv:17: assert: PFunction.cc:47: failed assertion statement_

```verilog
package test_pkg;

  virtual class uvm_void;
  endclass : uvm_void

  class uvm_object extends uvm_void;
    function new (string name = "uvm_object");
    endfunction : new 

    virtual function void print ();
      $display ("Print");
    endfunction : print 
  endclass : uvm_object

  class uvm_report_object extends uvm_object;
    function new (string name = "uvm_report_object");
      super.new (name);
    endfunction : new 
  endclass : uvm_report_object 

endpackage : test_pkg

module m;
  import test_pkg::*;
  uvm_object u0;
  
   initial begin : test
     #100;
     $display ("Hello World");
     u0 = new ();
     u0.print();

   end : test

endmodule : m

```

Thanks
Srini

steveicarus, 2020-11-22:
OK, this issue is not the string arguments, but the empty constructor uvm_report_object::new. (super.new is a special case.) I tested this assertion by putting a $display in the uvm_report_object::new, and it compiles and runs correctly. So I'm on top of it. Thanks for the report.

martinwhitaker, 2020-11-22:
See also issue #387.

steveicarus, 2020-11-22:
OK, I'll take on #387 as well. I see a couple bugs related to handling of super.new, and I have a holiday weekend coming up to work on it.

steveicarus, 2020-11-23:
Fixed here: https://github.com/steveicarus/iverilog/pull/394

closed 2020-11-23 by steveicarus