steveicarus/iverilog #391: SV: error in tasks with class argument

svenka3, 2020-11-22:
IVL 12.0 Devel version.

Error:
> ../ivl_i391.sv:41: sorry: I do not know how to elaborate r-value as IVL_VT_CLASS.

Looks like task run_phase (uvm_phase phase) causes issue - functions seem to work, tasks without class argument also works.

Thanks
Srini

Test:

```verilog
package test_pkg;

  virtual class uvm_void;
    function new ();
    endfunction : new 
  endclass : uvm_void

  class uvm_object extends uvm_void;
    function new ();
    endfunction : new 

    virtual function void print ();
      $display ("Print");
    endfunction : print 
  endclass : uvm_object

  class uvm_phase extends uvm_object;
    function new ();
    endfunction : new 
  endclass : uvm_phase 

  class uvm_report_object extends uvm_object;
    function new ();
    endfunction : new 
  endclass : uvm_report_object 

  class uvm_component extends uvm_report_object;
    function new ();
    endfunction : new 

    virtual function void build_phase (uvm_phase phase);
    endfunction : build_phase 

    virtual task run_phase (uvm_phase phase);
    endtask : run_phase 

    virtual task run_all ();
      uvm_phase p0;

      this.build_phase (p0);
      this.run_phase (p0);
    endtask : run_all 

  endclass : uvm_component 

endpackage : test_pkg

module m;
  import test_pkg::*;
  uvm_object u0;
  
   initial begin : test
     #100;
     $display ("Hello World");
     u0 = new ();
     u0.print();

   end : test

endmodule : m

```

martinwhitaker, 2020-11-23:
It looks like the compiler doesn't yet support passing objects to tasks or functions.
Your call to build_phase is being optimised away because the function body is empty, which is why it appears to work. That should really happen to the call to run_phase too, but the compiler doesn't spot that one.

martinwhitaker, 2020-11-23:
N.B. A "sorry" message always indicates a feature that isn't yet supported which is why I label this as an enhancement, not a bug.

martinwhitaker, 2020-11-24:
This turned out to be an easy one. Now supported in both the master and v11 branches.

closed 2020-11-24 by martinwhitaker