steveicarus/iverilog #392: Tiny implementation of UVM to IVL - how to collaborate?

svenka3, 2020-11-22:
Hello,
  Not a typical "issue" per-se, more of discussion - I along with a few interns have a small code base (non-trivial) that attempts to implement tiny parts of UVM in Icarus (we named it as IVL_UVM). We tried few approaches - top-down, bottom-up, then finally decided to do it from scract with some code leveraged from Accellera UVM library. 

We will be adding the code to a separate git-repo shortly. I am seeing issues that occur only in a larger code base and not always in tiny/unit tests (Didn't try too hard, am sure it should be possible). I did submit few issues based on those - derived from larger code base. 

How do you suggest to move ahead in resolving issues from not-so-trivial tests/code base. 

Thanks
Srini

steveicarus, 2020-11-22:
Please move this discussion to the iverilog-devel mailing list. That's
the ideal place for discussions of this sort.

closed 2020-11-22 by steveicarus

svenka3, 2020-11-22:
Is that the one of source-forge?

steveicarus, 2020-11-22:
Yes.At this point, the only thing that sourceforge still hosts of Icarus
Verilog is the mailing list.