steveicarus/iverilog #393: $fgets doesn't accept string as first argument

oj-amazing, 2020-11-23:
Whenever I use $fgets with a string input, I see the following error: 
 - $fgets's first argument must be a reg.

Sample code:

```
module testbench;
  integer fileIn;
  string msg;
  initial begin
    fileIn = $fopen("fileIn", "r");
    if (!fileIn) $error("BAD FILE");
    $fgets(msg, fileIn);
    $display("%s", msg);
    $fclose(fileIn);
  end
endmodule
```
Does Icarus support $fgets with string input? The code seems to work when using a different simulator on EDAPlayground.

steveicarus, 2020-11-23:
It currently does not. The standard does not say whether it is valid or not, either. The description, in fact, only covers the case that the msg is a bit vector.

**21.3.4.2 Reading a line at a time**

_One line can be read from a file using $fgets. For example:
  integer code;
  code = $fgets ( str, fd );
reads characters from the file specified by fd into the variable str until str is filled, or a newline character
is read and transferred to str, or an EOF condition is encountered. If str is not an integral number of bytes
in length, the most significant partial byte is not used in order to determine the size._

oj-amazing, 2020-11-23:
That's fair. Adding string support would definitely be a quality-of-life enhancement. Maybe regex too, later down the line. For now I'll just use a Python wrapper.

closed 2020-11-23 by oj-amazing

caryr, 2020-11-23:
I'm not so certain this should not be allowed. I would assume an actual string is always an integral number of bytes and being auto sized would read until the newline. I'm not certain if we have support for actual strings in the VPI so this may not be possible to add without significant VPI work first. For example sscanf() clearly does say it can read from a string.
> $sscanf reads from the argument str , which may be an expression of integral, unpacked array of byte, or
> string data type.

steveicarus, 2020-11-23:
I didn't say it shouldn't be supported. The standard seems to leave it
open. The standard doesn't say it should be an error, so I think it
would be a useful extension. If we want to do this, we should tag it
as an enhancement, and not a bug.