steveicarus/iverilog #399: CI

martinwhitaker, 2020-11-29:
It turns out that Travis CI is no longer automatically free to use for OSS projects - see

https://blog.travis-ci.com/2020-11-02-travis-ci-new-billing
https://travis-ci.community/t/org-com-migration-unexpectedly-comes-with-a-plan-change-for-oss-what-exactly-is-the-new-deal/

We could try applying for OSS credits, but who knows how long that will last. One of our problems is that testing on Windows takes an inordinate amount of time, so uses a lot of credits.

An alternative is GitHub Actions, which is (currently) free for public repositories. I'll take a look at that.

umarcor, 2020-11-30:
To Do from #400:

> we need to compare against the appropriate expected results file (and generate version-specific expected results for MSYS2). Alternatively we could label all expected failures as such, and modify vvp_reg.pl to return a pass/fail exit code (as I did for vpi_reg.pl).

> Looks like we can create different scripts for different branches. We can use that to pass different expected results to the v11 branch.

steveicarus, 2020-11-30:
I agree with that approach. I was, in fact, thinking the same thing.

Make it so;-)

steveicarus, 2020-11-30:
Looks like we are switched over to Github Actions (PR #400). Can I close this issue?

umarcor, 2020-11-30:
@steveicarus, I'd rather keep it open until we implement the different tests depending on the branch. Anyway, feel free to close it.

umarcor, 2020-11-30:
It seems that #401 was executed in CI, merged, and executed in CI again, before latest changes were pushed to ivtest. As a result, although CI status is green, any CI run will fail due to missing files. See https://github.com/umarcor/iverilog/actions/runs/390884757.

I worked around it by fixing ivtest to an specific commit: https://github.com/umarcor/iverilog/runs/1472060125?check_suite_focus=true. Is that desired?

steveicarus, 2020-11-30:
I think, on the master branch, once a PR is merged, it doesn't matter. The test suite is going to advance as we add new fixes, so old commits will start to fail. I don't think it matters.

umarcor, 2020-11-30:
@steveicarus, please, restart https://github.com/steveicarus/iverilog/actions/runs/390778127 and you'll see what I mean. The first problem is `                   br_gh307: Failed - missing source file.`. That's because of the last commit in https://github.com/steveicarus/ivtest/commits/master.

steveicarus, 2020-11-30:
Fixed. You were correct that a file was missing.

martinwhitaker, 2020-12-05:
I've back-ported the CI changes to the v11 branch. I modified the ivtest script that generates the expected results for MSYS2 to support different versions. I've deleted the saved copy of the results in ivtest - the CI automatically regenerates them.

closed 2020-12-05 by martinwhitaker