steveicarus/iverilog #409: Running simulation with Icarus remains hanging

rryantt, 2020-12-05:
I have an n-bit ripple adder circuit and I have been playing around with the design module because I am new to verilog. However, I upon making a slight modification to the code, icarus is unable to run the simulation and remains hanging. I am wondering why this is a problem. Below is the code. I observed the error is from this line `fulladder ad(cin[i], carry_in, c_in[25], cin[i+1], sum[i]); `, specifically the presence of `cin[i]` and `cin[i+1]`. Since these are inputs why is the simulator hanging.

```
`include "fulladder.sv" 
module rippleadder #(parameter width = 64)(input [width-1:0] a, b, input carryin, output logic carryout, output [width-1:0] sum);
logic [width:0] cin; 
 genvar i; 
always @(*) begin 
 cin[0] =carry_in; 
 carry_out =cin[width]; 
 end 
generate 
 for(i=0; i<width; i=i+1) begin 
fulladder ad(c_in[i], carry_in, c_in[25], c_in[i+1], sum[i]); 
 end 
 endgenerate 
endmodule

```

caryr, 2020-12-07:
We need a fully functional test example. The simulator hanging is usually a race in the simulation and is very common when learning Verilog. We also like to see a reduced example if possible and often you discover the issue while reducing the code down (e.g. is it width dependent, does it happen with a single instance, etc.). This might seem like extra work, but it is really learning how to debug issues and you learn a lot about a language by struggling with it.

We can provide help as needed. The first thing to look at is does -pfileline=1 provide a clue if procedural lines are interacting badly. This will generate a message to the output as each procedural line is executed. If you have written this using continuous assignments then it will probably not help, but it is worth checking.

rryantt, 2020-12-07:
@caryr  Thanks for the feedback. I have updated the code. I also adder the flag `-pfileline=1` to the command and this is output I got.
```
./rippleadder.sv:7: Blocking assignment.
./rippleadder.sv:5: Event wait (@) statement.
./rippleadder.sv:3: Blocking assignment.
./rippleadder:4.sv: Blocking assignment.
./rippleadder.sv:2: Event wait (@) statement.
./rippleadder.sv:3: Blocking assignment.
./rippleadder.sv:4: Blocking assignment.
./rippleadder.sv:2: Event wait (@) statement.
./rippleaddersv:3: Blocking assignment.
./rippleadder.sv:4: Blocking assignment.
./rippleadder.sv:2: Event wait (@) statement.
./rippleadder.sv:3: Blocking assignment.
./rippleadder.sv:4: Blocking assignment.
...
```
It continues in this manner in an infinitely

martinwhitaker, 2020-12-07:
Consider the code your generate loop generates for i=24. Can you now see the infinite loop you've created?

caryr, 2021-01-20:
Closing as invalid

closed 2021-01-20 by caryr