steveicarus/iverilog #410: IVL_UVM: internal error around class method calls

svenka3, 2020-12-05:
Hi,
  I have managed to reproduce one of the crashes from Icarus (12.0 dev release) that has been bothering me for a week or so using a single file. It is not trivial code, so uploaded to:

https://github.com/svenka3/code_to_ivl_team/blob/main/ivl_uvm_comp_crash/ivl_uvm_single.sv

(Than pasting code below).

I get a crash as below:

```log
../ivl_uvm_single_file.sv:999: warning: User function 'm_check_verbosity' is being called as a task.
../ivl_uvm_single_file.sv:1091: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1210: internal error: Can't find task uvm_count_info in class uvm_component
../ivl_uvm_single_file.sv:1210: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1214: internal error: Can't find task uvm_count_info in class uvm_component
../ivl_uvm_single_file.sv:1214: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1218: internal error: Can't find task uvm_count_info in class uvm_component
../ivl_uvm_single_file.sv:1218: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1221: internal error: Can't find task uvm_count_info in class uvm_component
../ivl_uvm_single_file.sv:1221: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1226: internal error: Can't find task uvm_count_info in class uvm_component
../ivl_uvm_single_file.sv:1226: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1184: internal error: Can't find task uvm_count_info in class uvm_object
../ivl_uvm_single_file.sv:1184: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1291: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1292: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1293: warning: User function 'uvm_count_info' is being called as a task.
../ivl_uvm_single_file.sv:1294: warning: User function 'uvm_count_info' is being called as a task.
6 error(s) during elaboration.

```

martinwhitaker, 2020-12-06:
You really need to give us a simpler test case. Try removing a bit at a time until you get to the minimum amount of code that demonstrates the issue.

svenka3, 2020-12-06:
Agree with you Martin. (I started doing that and have found one another issue while doing so, filed just now).

 This is the challenge I am facing - the same code worked fine 2 weeks back and with some of the fixes (guess around the super.new/task arguments etc.) it broke. 

I am open to ideas on how to add this code base to your regression suite once we get it working again. My goal is to slowly build UVM pieces that run on Icarus, so we will need some mechanism to track and keep the sanity intact. I will add to the sv-tests repo in near future if that would help your dev team.

Thanks for your patience.

caryr, 2020-12-07:
We probably need to discuss how best to add this to ivtest. I assume this will match UVM as a library so I believe the best way to do this is to create a subdirectory for the library part and then build individual tests to check the various components. These individual tests can include a toplevel file that loads all the library components. If you look at our current tests you should get an idea about how we break things down into pieces and the style of the checks (e.g. using self checking where possible and printing "PASSED" when everything works as expected). We typically discuss these kind of things on the iverilog-devel mailing list on SourceForge.

svenka3, 2020-12-07:
Thnks Cray for detailed comments. Let me think about it and come back. Yes tests can be very modular and tiny, infact UVM development follows the same for testing the code. But the challenge here is the library itself. 

Am sure we can handle , let me get it running in a version

caryr, 2021-02-17:
Please check if the changes I submitted today resolve this.

svenka3, 2021-02-20:
Internal errors are replaced with "error" - I still believe scoping of package level functions do not seem to work. Would you like a smaller test?

caryr, 2021-02-21:
Simple tests are always appreciated

svenka3, 2021-02-21:
I tried reducing it. I have similar/same error - but perhaps it is to do with package scope resolution.

```verilog
package p;

  function void uvm_report_info (string s = "s");
    $display ("%m %s", s);
  endfunction : uvm_report_info 

  class uvm_object;
    function new();
      uvm_report_info ("obj");
    endfunction : new
  endclass

endpackage : p

module m;
  import p::*;
endmodule : m

```

svenka3, 2021-06-06:
Just checking if any of the developers had time to look at this test. I was hoping some of the scope resolution related fixes done few months back would fix this. 

Any comments please?

Thanks

PR #784 (2022-10-17): Handle calling void function from class method
Calling a void function (or a regular function and ignoring the result)
from within a class method will currently result in an error unless the
void function itself is a method of the same class.

This is because we add the implicit `this` as an object on which to search
for the function and if we do not find it print an error. Change this to
only print an error if the implicit this was not added and it was a method
call on an object identifier.

This resolves #410 and #419.
closed 2022-10-17 by steveicarus