steveicarus/iverilog #411: SV:  assert: elaborate.cc:5299: failed assertion statement

svenka3, 2020-12-06:
Hi,
  As I started extracting a smaller test from IVL_UVM crash reported earlier, below is a smaller test and shows an error as:

```log
iverilog -DVW_IVLOG_GO2UVM -DIVL_UVM -g2012 -I../ -o vw_ivl_go2uvm.vvp ../ivl_uvm_msg.svh

 ../ivl_uvm_msg.svh:79: assert: elaborate.cc:5299: failed assertion statement_
```

Code below.

Thanks
Srini

```verilog

`define uvm_info(ID, MSG, VERBOSITY) 
 
 `define g2u_display(MSG) \
   begin \
     $display (MSG); \
   end

package test_pkg;

  int uvm_info_counter;
  int uvm_warn_counter;
  int uvm_err_counter;
  int uvm_fatal_counter;

  function void uvm_count_info();
    uvm_info_counter++;
  endfunction : uvm_count_info

  /*

  function void uvm_count_info();
    uvm_info_counter++;
  endfunction : uvm_count_info
  */

  function void uvm_count_warn();
    uvm_warn_counter++;
  endfunction : uvm_count_warn

  function void uvm_count_err();
    uvm_err_counter++;
  endfunction : uvm_count_err

  function void uvm_count_fatal();
    uvm_fatal_counter++;
  endfunction : uvm_count_fatal

  function void report_summarize ();
    int num_errs;

    $display("");
    $display("--- UVM Report Summary ---");
    $display("");
    $display("** Report counts by severity");
    $display ("UVM_INFO : %0d", uvm_info_counter);
    $display ("UVM_WARNING : %0d", uvm_warn_counter);
    $display ("UVM_ERROR : %0d", uvm_err_counter);
    $display ("UVM_FATAL : %0d", uvm_fatal_counter);

    num_errs = uvm_err_counter;

    if(num_errs > 0) begin : fail
      $display ( "%c[1;31m",27 ) ; // RED color
      $display ("Test FAILED with %0d error(s), look for UVM_ERROR in log file",
                 num_errs);
      $display ( "%c[0m",27 ) ;
    end : fail
    else begin : pass
      $display ( "%c[5;34m",27 ) ; // BLUE color
      $display ( "*** Congratulations! Test PASSED with NO UVM_ERRORs ***" ) ;
      $display ( "%c[0m",27 ) ;
    end : pass

  endfunction : report_summarize

virtual class uvm_void;
  function new ();
  endfunction : new 
endclass : uvm_void

class uvm_object extends uvm_void;
  function new ();
    super.new ();
  endfunction : new 

  virtual function void print ();
    `uvm_info (log_id, "Print", UVM_MEDIUM);
  endfunction : print 

endclass : uvm_object

class uvm_report_object extends uvm_object;
  function new ();
    super.new ();
  endfunction : new 

endclass : uvm_report_object 

class uvm_phase extends uvm_object;
  function new ();
    super.new ();
  endfunction : new 

endclass : uvm_phase 

virtual class uvm_component extends uvm_report_object;
  function new ();
    super.new ();
  endfunction : new 

  virtual function void build_phase(uvm_phase phase);
    `g2u_display ("build_phase")
  endfunction : build_phase

  virtual function void connect_phase(uvm_phase phase);
    `g2u_display ("connect_phase")
  endfunction : connect_phase

  virtual function void end_of_elaboration_phase(uvm_phase phase);
    `g2u_display ("end_of_elaboration_phase")
  endfunction : end_of_elaboration_phase
  virtual function void start_of_simulation_phase(uvm_phase phase);
    `g2u_display ("start_of_simulation_phase")
  endfunction : start_of_simulation_phase

  //virtual task run_phase (uvm_phase phase);
  virtual task run_phase ();
    `g2u_display ("run_phase")
    this.print ();
  endtask : run_phase 

  virtual function void extract_phase(uvm_phase phase);
  endfunction : extract_phase
  virtual function void check_phase(uvm_phase phase);
  endfunction : check_phase
  virtual function void report_phase(uvm_phase phase);
  endfunction : report_phase
  virtual function void final_phase(uvm_phase phase);
  endfunction : final_phase

  virtual task ivl_uvm_run_all_phases ();
    uvm_phase u_ph_0;

    u_ph_0 = new();
    this.build_phase (u_ph_0);
    this.connect_phase (u_ph_0);
    this.end_of_elaboration_phase (u_ph_0);
    this.start_of_simulation_phase (u_ph_0);
    this.run_phase ();
    this.extract_phase (u_ph_0);
    this.check_phase (u_ph_0);
    this.report_phase (u_ph_0);
    this.final_phase (u_ph_0);
  endtask : ivl_uvm_run_all_phases 

endclass : uvm_component

endpackage : test_pkg

module m;

  import test_pkg::*;

endmodule : m

```

martinwhitaker, 2020-12-07:
Here is a fully cut-down test case:
````
module test();

function void do_nothing();
  ;
endfunction

initial do_nothing();

endmodule
````

svenka3, 2020-12-07:
Thanks a lot Martin for the reduction, will attempt it myself next time onwards

martinwhitaker, 2020-12-10:
Fixed in both the master and v11 branches.

closed 2020-12-10 by martinwhitaker