steveicarus/iverilog #412: Q.size() in wait() -->   Assertion `nex' failed

svenka3, 2020-12-06:
Hi,
  Below code gives an assertion error as:

```log
iverilog -DVW_IVLOG_GO2UVM -DIVL_UVM -g2012 -I../ -o vw_ivl_go2uvm.vvp  -f flist.class 2>&1 | tee go2uvm_comp.log 	
ivl: t-dll-proc.cc:923: virtual bool dll_target::proc_wait(const NetEvWait*): Assertion `nex' failed.

```

Code:

```verilog
// Mailbox via Q
//

module ivl_uvm_mbx; // #(parameter T = 3) ();

  int mbx_via_q [$];
  int q_size;

  task put (input int ival);
    mbx_via_q.push_back (ival);
    q_size = mbx_via_q.size();
  endtask : put 

  task get (output int oval);
    wait (mbx_via_q.size() > 0);
    wait (q_size > 0);
    $display ("MBX: Size: %0d", mbx_via_q.size());
    oval = mbx_via_q.pop_front ();
    q_size = mbx_via_q.size();
    $display ("MBX: Size: %0d", mbx_via_q.size());
  endtask : get 

endmodule : ivl_uvm_mbx 

module m;

  ivl_uvm_mbx mbx_0 ();

  int pval,gval;

  initial begin
    pval = $random();
    mbx_0.put(pval);
    mbx_0.get(gval);
    $display ("put: 0x%0h get: 0x%0h", pval, gval);
  end
endmodule : m

```

caryr, 2020-12-07:
When we request people create reduced examples we are looking for something like the following not what was provided above:

```
module top;
  int mbx[$];

  initial begin
    mbx.push_back(1);
    wait(mbx.size());
    $display("PASSED");
  end
endmodule
```

Currently all work on Icarus is volunteer time so we would rather spend out time figuring out and fixing issues not reducing the example before working on a solution. Some of your other reports are very complicated so reducing them down could take us significant time since we are not familiar with your intention so we need to learn the code and then reduce. This one was fairly easy because you had correctly identified that it was the queue size() method that was causing issues.

svenka3, 2020-12-07:
Sure, appreciate it. I do sincerely acknowledge the hard work out in by your team. Just FYI, my work on IVL_UVM is also on a voluntary basis so I definitely understand what you are saying here.

Cheers

commit fe9873c2 (2020-12-14): Handle dynamic queue objects in event context

In the rare cases where DARRAY signals are in the network, handle
them by creating the proper ivl_nexus_t node. This also implements
the receive of vvp_object_t objects bu vvp_fun_anyedge_sa. This
together makes it possible for IVL_VT_DQUEUE objects to be in
wait lists.

This fixes #412
commit 788664a5 (2020-12-14): Handle dynamic queue objects in event context

In the rare cases where DARRAY signals are in the network, handle
them by creating the proper ivl_nexus_t node. This also implements
the receive of vvp_object_t objects bu vvp_fun_anyedge_sa. This
together makes it possible for IVL_VT_DQUEUE objects to be in
wait lists.

This fixes #412
commit e5f68d7c (2020-12-14): Handle dynamic queue objects in event context

In the rare cases where DARRAY signals are in the network, handle
them by creating the proper ivl_nexus_t node. This also implements
the receive of vvp_object_t objects bu vvp_fun_anyedge_sa. This
together makes it possible for IVL_VT_DQUEUE objects to be in
wait lists.

This fixes #412
PR #424 (2020-12-14): Handle dynamic queue objects in event context
In the rare cases where DARRAY signals are in the network, handle
them by creating the proper ivl_nexus_t node. This also implements
the receive of vvp_object_t objects bu vvp_fun_anyedge_sa. This
together makes it possible for IVL_VT_DQUEUE objects to be in
wait lists.

This fixes #412
closed 2020-12-14 by steveicarus