steveicarus/iverilog #415: SV: $sformatf format specifier incorrect - additional check needed

svenka3, 2020-12-12:
Hi,
  Please see below code. Issue specifically is around $sformatf ("0x%0h", class_obj)

`DV_CHECK_NE(1, 0) --> works
    `DV_CHECK_NE(c_0, null) --> Fails with below error:

```log
iverilog devel version:

iverilog -DVW_IVLOG_GO2UVM -DIVL_UVM -g2012 -I../ -o vw_ivl_go2uvm.vvp  -f flist.class 2>&1 | tee go2uvm_comp.log 	
ivl: draw_vpi.c:430: draw_vpi_taskfunc_args: Assertion `0' failed.
sh: line 1: 18596 Done                    /usr/local/lib/ivl/ivlpp -L -F"/tmp/ivrlg22ab1ed4d" -f"/tmp/ivrlg2ab1ed4d" -p"/tmp/ivrli2ab1ed4d"
     18597 Aborted                 (core dumped) | /usr/local/lib/ivl/ivl -C"/tmp/ivrlh2ab1ed4d" -C"/usr/local/lib/ivl/vvp.conf" -- -
vvp -l go2uvm_run.log vw_ivl_go2uvm.vvp
vw_ivl_go2uvm.vvp:29: syntax error
```

Extracted test - 

```verilog

`define uvm_info(ID, MSG, VERB) \ 
   begin \
     $display (MSG); \
   end

`define DV_CHECK_NE(ACT_, EXP_, SEV_=info) \
    begin \
        `uvm_``SEV_(ID_, $sformatf("Check failed 0x%0h vs 0x%0h) ", \
                                     ACT_, EXP_ ), 10) \
    end \

class foo;
endclass : foo

module m;

  foo c_0;
  foo c_1;

  initial begin
    c_0 = new();

    `DV_CHECK_NE(1, 0)
    `DV_CHECK_NE(1, 1)
    `DV_CHECK_NE(c_0, null)
    //`DV_CHECK_NE(c_1, null)
  end

endmodule : m

```

svenka3, 2020-12-12:
Just to note - other tools do allow with a warning at times. But I believe the code is wrong - with %h it uses a class_handle, a better error messaging may be a first fix. If you can still print the address (of the obj) or something - that maybe better (to match with other tools).

martinwhitaker, 2020-12-13:
Simplified test case:
````
module test();

initial $display("%h", null);

endmodule
````

caryr, 2020-12-14:
There is also the issue with $display() not supporting printing of a class object. I'll take a look at these issues.

caryr, 2020-12-14:
Here is a relevant part of the standard regarding handling of "null":

> An argument consisting of the special value null shall be represented as an expression with a vpiType of vpiConstant and a vpiConstType of vpiNullConst.

caryr, 2020-12-14:
I have added support for handling null object in the display tasks so this should no longer fail for the example Martin provided (it now prints 'null'). I am still working on support for printing something usable for class variables.