steveicarus/iverilog #418: SV: void function leads to assert: elaborate.cc:3873:

svenka3, 2020-12-13:
Hi,
  I am getting below error if I declare a void function (that gets called under another function).

Run as:
> 	iverilog -DVW_IVLOG_GO2UVM -DBUG -DIVL_UVM -g2012 -I../ -o vw_ivl_go2uvm.vvp  -f flist.class 2>&1 | tee go2uvm_comp.log 	

```log
Iverilog devel version
iverilog -DVW_IVLOG_GO2UVM -DBUG -DIVL_UVM -g2012 -I../ -o vw_ivl_go2uvm.vvp  -f flist.class 2>&1 | tee go2uvm_comp.log 	
../ivl_uvm_sf.sv:26: assert: elaborate.cc:3873: failed assertion dscope->elab_stage() >= 3
sh: line 1: 32154 Done                    /usr/local/lib/ivl/ivlpp -L -F"/tmp/ivrlg25fd99de1" -f"/tmp/ivrlg5fd99de1" -p"/tmp/ivrli5fd99de1"
     32155 Aborted                 (core dumped) | /usr/local/lib/ivl/ivl -C"/tmp/ivrlh5fd99de1" -C"/usr/local/lib/ivl/vvp.conf" -- -
```

Test below, look for BUG, when you define you get this error, else not. 

```verilog
`define IVL_UVM

typedef bit [1:0] uvm_severity;
parameter UVM_INFO = 0;

package ivl_uvm_pkg;

  function int uvm_report_enabled(int verbosity, 
                          uvm_severity severity=UVM_INFO, string id="");
    if (get_report_verbosity_level(severity, id) < verbosity) begin
      return 0;
    end else begin
      return 1;
    end
  endfunction : uvm_report_enabled

  `ifdef BUG
  function void m_check_verbosity();
  `else
  function m_check_verbosity();
  `endif
  endfunction : m_check_verbosity

  function int get_report_verbosity_level(uvm_severity severity=UVM_INFO, string id="");
    m_check_verbosity ();
    return 1;
  endfunction : get_report_verbosity_level

endpackage : ivl_uvm_pkg

module m;
  import ivl_uvm_pkg::*;

endmodule : m

```

martinwhitaker, 2020-12-13:
Simplified test case:
````
module m;

  function void funct1();
  endfunction

  function void func2();
    funct1();
  endfunction

endmodule
````
The bug is only triggered if the name of the first function contains more characters than the name of the second function.

martinwhitaker, 2020-12-13:
Fixed in both the master and v11 branches.

closed 2020-12-13 by martinwhitaker

caryr, 2020-12-26:
I just looked at the test in the testsuite and is it intended that both paths display "PASSED"? I assume the second one should be FAILED.

martinwhitaker, 2020-12-26:
Yes, an oversight on my part, mainly because the bug was a compiler assertion, so never got as far as running the test. Fixed now.