steveicarus/iverilog #419: SV: function declared after its call leads to internal error (IVL_UVM reduced test)

svenka3, 2020-12-13:
Hi,
  Below is a reduced test from recent crash I reported on IVL_UVM. Hope this is small enough for developers to debug and fix.

Thanks
Srini

```verilog
`define IVL_UVM

package ivl_uvm_pkg;

class uvm_phase;
endclass : uvm_phase 

virtual class uvm_component;
  function new ();
  endfunction : new 

  virtual function void end_of_elaboration_phase(uvm_phase phase);
    uvm_info_counter ();
  endfunction : end_of_elaboration_phase
endclass : uvm_component

  function uvm_count_info();
    uvm_info_counter++;
  endfunction : uvm_count_info

int uvm_info_counter;
string log_id = "IVL_GO2UVM";

endpackage : ivl_uvm_pkg

module m;
  import ivl_uvm_pkg::*;

  uvm_component c_0;

  initial begin
    c_0 = new ("c_0");
  end

endmodule : m

```

```log
iverilog -DVW_IVLOG_GO2UVM -DIVL_UVM -g2012 -I../ -o vw_ivl_go2uvm.vvp  -f flist.class 2>&1 | tee go2uvm_comp.log 	
../ivl_uvm_sf.sv:14: internal error: Can't find task uvm_info_counter in class uvm_component
../ivl_uvm_sf.sv:14: error: Enable of unknown task ``uvm_info_counter''.
../ivl_uvm_sf.sv:14: error: Unable to elaborate statement in function end_of_elaboration_phase.
```

martinwhitaker, 2020-12-13:
I think the only issue here is that it is being reported as an internal error. It is a genuine error in the code being compiled - there is no task or method named uvm_info_counter. I've changed the compiler error message accordingly.

svenka3, 2020-12-15:
Martin,
  That was a typo in code, I just now tried with uvm_count_info() as the function call, yet we get the same error. Testcase and log below.

```log
../ivl_uvm_i419.sv:18: internal error: Can't find task uvm_count_info in class uvm_component

```

```verilog

`define IVL_UVM

package ivl_uvm_pkg;

  function uvm_count_info();
    uvm_info_counter++;
  endfunction : uvm_count_info
class uvm_phase;
endclass : uvm_phase 

virtual class uvm_component;
  function new ();
  endfunction : new 

  virtual function void end_of_elaboration_phase(uvm_phase phase);
    uvm_count_info ();
  endfunction : end_of_elaboration_phase
endclass : uvm_component

int uvm_info_counter;
string log_id = "IVL_GO2UVM";

endpackage : ivl_uvm_pkg

module m;
  import ivl_uvm_pkg::*;

  uvm_component c_0;

  initial begin
    c_0 = new ();
  end

endmodule : m
```

svenka3, 2020-12-15:
It looks like an issue with package level function not visible inside a class (class is also inside same package). In the larger test, I duplicated this function in a base class and this error goes away.

Thanks

martinwhitaker, 2020-12-15:
OK, that's a genuine bug. May be related to issue #292. I found quite a few things wrong with the way the compiler currently performs symbol searches.

caryr, 2021-02-17:
Does the declaration for uvm_info_counter need to be before the function uvm_count_info?

martinwhitaker, 2024-01-09:
This was fixed by PR #784.

closed 2024-01-09 by martinwhitaker