steveicarus/iverilog #421: SV: abstract class not working as expected

svenka3, 2020-12-13:
Hi,
  Abstract class handle allows construction to an object, see below code. Is this check not implemented in Icarus yet?

Thanks
Srini

```verilog
package test_pkg;

  virtual class ivl_uvm_base_test;
    function new ();
    endfunction : new 

    virtual task run_phase ();
      $display ("%m run_phase");
    endtask : run_phase 
  endclass : ivl_uvm_base_test

  task run_test ();
    ivl_uvm_base_test u_t_0;

    u_t_0 = new();
    u_t_0.run_phase ();
  endtask : run_test 
endpackage : test_pkg

module m;
  import test_pkg::*;

  initial begin
    run_test ();
  end
endmodule : m

```

martinwhitaker, 2020-12-13:
Using the master branch, I get:
````
% iverilog -g2009 ivltests/br_gh421.v
% a.out
test_pkg.ivl_uvm_base_test.run_phase run_phase
````
What are you expecting?

svenka3, 2020-12-14:
Thanks for the response, as per LRM this should throw an error as it is an abstract class and not a concrete one.

martinwhitaker, 2020-12-14:
There is currently no support for abstract classes. The virtual keyword is accepted by the parser, but otherwise ignored. This is also true for virtual methods.

PR #815 (2022-12-17): Do not allow to create objects of virtual classes
It is not allowed to create objects of virtual classes. Currently the
virtual keyword is accepted by the parser, but otherwise ignored.

Keep track of whether a class is virtual and generate an error when the
class new operator is used for a virtual type.

Resolves #421.
closed 2022-12-17 by steveicarus