steveicarus/iverilog #422: SV: issue with type compatibility across inheritence tree

svenka3, 2020-12-13:
Hi,
  In basic OOP using SV, base_handle = der_handle should work. But below code gives error as follows:

```log
../ivl_uvm_vir_meth.sv:23: error: the type of the variable 'u_t_0' doesn't match the context type.
../ivl_uvm_vir_meth.sv:23:      : variable type=class my_test{}
../ivl_uvm_vir_meth.sv:23:      : context type=class ivl_uvm_base_test{}
```

Is this not supported yet? If so, can you please prioritize this? This would help basic UVM structure/test infrastructure. 

Thanks
Srini

Code;
```verilog
package test_pkg;

  virtual class ivl_uvm_base_test;
    function new ();
    endfunction : new 

    virtual task run_phase ();
      $display ("%m run_phase");
    endtask : run_phase 
  endclass : ivl_uvm_base_test

  class my_test extends ivl_uvm_base_test;
    virtual task run_phase ();
      $display ("%m run_phase");
    endtask : run_phase 
  endclass : my_test 

  ivl_uvm_base_test uvm_test_top;

  task run_test ();
    my_test u_t_0;

    uvm_test_top = u_t_0;

    u_t_0 = new();
    u_t_0.run_phase ();
  endtask : run_test 
endpackage : test_pkg

module m;
  import test_pkg::*;

  initial begin
    run_test ();
  end
endmodule : m

```

martinwhitaker, 2020-12-13:
Steve can correct me if I'm wrong, but I don't think there is any support for polymorphism as yet.

caryr, 2026-11-08:
After #1724 this compiles and runs. Please double check that this new functionality works as you expect.

closed 2026-11-08 by caryr

svenka3, 2026-11-09:
Looks good to me, motivated me to revamp ivl_uvm project. Thanks!