steveicarus/iverilog #433: SV: Q pop_back with return value ignored gives error

svenka3, 2020-12-20:
```log
iverilog 12.0 devel version
bb.sv:10: error: Enable of unknown task ``__current_scopes.pop_back''.
```

Code

```verilog

module m;
  int __current_scopes [$];
  int a;

  initial begin
    __current_scopes.push_front(10);
    // WORKS 
    // a = __current_scopes.pop_back(); 
    __current_scopes.pop_back(); 
  end

endmodule : m

```

martinwhitaker, 2020-12-20:
This should be accepted but generate a warning.

caryr, 2020-12-20:
It looks as if we never added functionality for calling method functions in a void context. Only user defined functions currently have this functionality. FYI you can replace the  `pop_back()` with the following code to make progress while this is being fixed.
```
q.delete(q.size()-1);
```

caryr, 2020-12-21:
This is now fixed in both v11 and devel

closed 2020-12-21 by caryr