steveicarus/iverilog #435: SV: comment inside macro leads to syntax error

svenka3, 2020-12-20:
I believe comments inside a fancy text macro with arguments 9that expands to multiple lines) is a vague area in LRM. However, UVM library uses this "feature" and Icarus gives a compile error. Below is a reduced test.

Thanks
Srini

Line 3 below, i.e. 2nd line inside the macro leads to a compile error.

```verilog
`define uvm_field_utils_end \
     if(what__ inside {UVM_SETINT,UVM_SETSTR,UVM_SETOBJ}) begin \
        // remove all scopes recorded (through super and other objects visited before) \
          __current_scopes.pop_back()); \
        __m_uvm_status_container.m_uvm_cycle_scopes = __current_scopes; \
     end \

module m;
endmodule : m

```

martinwhitaker, 2020-12-20:
Although the standard isn't explicit, I think the intent is clear - the comment should terminate at the '\'.

martinwhitaker, 2020-12-20:
I'll take this one on.

caryr, 2020-12-20:
FYI single line comments embedded in a macro used to fail in VCS, but I checked and that is no longer true. The argument at the time was the single line comment ends at the newline which technically eats up the \. You could do /* */ \ as long as it was a single line, but even that is no longer the case. For the /* */ comment case it now eats up everything between the /* and */ and only requires a \ after the closing */

martinwhitaker, 2020-12-20:
Fixed in both master and v11 branches.

closed 2020-12-20 by martinwhitaker