steveicarus/iverilog #436: SV: string len() operator in expressions not working

svenka3, 2020-12-20:
Hi,
  A string Q is used, its elements are processed inside a foreach, so far it works. Now, foreach element in this string Q, say we want to compare the length of the string and do some processing, Icarus breaks. 

Minmalist test:

```verilog
module m;

  string m_argv [$];

  initial begin
    m_argv.push_back ("str0");

    foreach (m_argv[i]) begin
      if(m_argv[i].len() >= 2) begin
        $display ("LARGE: %0d", lv_len);
      end
  
    end
  end
endmodule : m

```

```log
bb.sv:9: error: Scope index expression is not constant: i
bb.sv:9: error: Scope index expression is not constant: i
bb.sv:9: error: No function named `m_argv[i].len' found in this context (m.$ivl_foreach0).
bb.sv:9: error: Unable to elaborate condition expression.
4 error(s) during elaboration.
```

I tried breaking the code to get individual string, get it length, compare, that works as below.

```verilog
module m;

  string m_argv [$];
  string lv_s;
  int lv_len;

  initial begin
    m_argv.push_back ("str0");
    m_argv.push_back ("str1");
    m_argv.push_back ("str0123");

    foreach (m_argv[i]) begin
      $display ("%s", m_argv[i]);
      lv_s = m_argv[i];
      lv_len = lv_s.len();
      $display ("%0d", lv_len);
      if(lv_len >= 2) begin
        $display ("LARGE: %0d", lv_len);
      end
   
    end
  end
endmodule : m

```

caryr, 2020-12-21:
This all looks to be in code that Steve might be working on. I have the start of a patch, but for now I am releasing this. The fundamental issue is that the function call code does not know how to correctly elaborate indexed operations like identifiers can. I was able to get the test_width routines to work correctly since we just need to know the underlying array type (string in this case), but getting a NetESelect added to the function call elaboration is not obvious and I don't want to do a bunch of hacking if this is being reworked.

steveicarus, 2020-12-31:
I'll give this a try. It may be a big job, as it seems to be uncovering a few problems with name resolution in l-values.

caryr, 2020-12-31:
Do you want the work I already started or are you going to roll your own?

steveicarus, 2020-12-31:
What have you done? I'm first cleaning up some weird behaviors (i.e. strange type elaborations) Could you at least show me what you've done? Maybe I'll steal it.

caryr, 2020-12-31:
I'll send you the patch in email.

closed 2021-01-18 by steveicarus