steveicarus/iverilog #437: SV: Multiple packages leads to crash

svenka3, 2020-12-23:
Hi,
  Using iverilog 12.0 devel, latest build. If I extend a class from one package in another I get a crash. Code and log below. If I move them to the same package it works. 

Thanks
Srini

```verilog
package ivl_uvm_pkg;
  virtual class uvm_test;
  endclass : uvm_test 
endpackage : ivl_uvm_pkg

package test_pkg;
  import ivl_uvm_pkg::*;
  class sanity_test extends uvm_test;
  endclass : sanity_test 
endpackage : test_pkg

module m;

  import ivl_uvm_pkg::*;
  import test_pkg::*;

endmodule : m

```
Log:
```log
iverilog -g2012 ivl_uvm_i436.sv
sh: line 1: 28072 Done                    /usr/local/lib/ivl/ivlpp -L -F"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg21c1ead4a" -f"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg1c1ead4a" -p"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrli1c1ead4a"
     28073 Segmentation fault: 11  | /usr/local/lib/ivl/ivl -C"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlh1c1ead4a" -C"/usr/local/lib/ivl/vvp.conf" -- -
```

martinwhitaker, 2020-12-23:
Fixed in both master and v11 branch.

closed 2020-12-23 by martinwhitaker

svenka3, 2020-12-24:
Fantastic turn-around Martin et al. This would help me use UVM_TESTNAME define and demo various tests as in UVM.

Thanks