steveicarus/iverilog #439: SV: built-in method srandom not supported

svenka3, 2020-12-27:
Hi,
  I understand random gen is NYI (Not Yet Implemented). However, from UVM lib compile/parse perspective it would help to get parser accept random() call. Below is a test:

```verilog
package t;

  bit use_uvm_seeding;

  class c;
    function void reseed ();
      if(use_uvm_seeding)
	this.srandom(1);
  endfunction

endclass : c
endpackage

module m;
  import t::*;

  initial begin
  end
endmodule 

```

```log
aa.sv:9: error: Can't find task srandom in class c
aa.sv:9: error: Enable of unknown task ``@.srandom''.
2 error(s) during elaboration.
```

caryr, 2020-12-27:
Based on your comments I assume you want this to be parsed and turned into a NOP or are you looking for a parse and make it a run-time error if executed? The later seems safer. This also should have been added to #369 to keep everything in one place.

I assume you understand adding deterministic randomization and the required constraints is likely a very large task.

svenka3, 2020-12-27:
NOP would be OK as we know constraint random generation is a big feature and not available in Icarus (srandom is a small part of that, not widely used per-se, also not recommended, but UVM has some calls to it).

Maybe a warning if you care to, but not very useful IMHO.

Thanks

svenka3, 2026-10-11:
Picking this up. Confirming the approach before sending a PR.

  **Proposed fix:**

  `this.srandom(seed)` — intercept at elaboration, emit a compile-time warning,
  treat as a NOP. Compilation succeeds; the warning informs the user the call has
  no effect.

  @caryr — does a warning + NOP approach sound OK?