steveicarus/iverilog #440: SV: Evaluate VEC4 expression (22) - error on wrong code

svenka3, 2020-12-27:
Iverilog devel version, I get 

```log
XXXX Evaluate VEC4 expression (22)
```

Code below. Function create() has wrong return value leading to this.

```verilog
package t;

  class c;
    virtual function create (string name=""); 
    return null; 
  endfunction

endclass : c
endpackage

module m;
  import t::*;

  initial begin
  end
endmodule 

```

caryr, 2020-12-28:
This now correctly reports the incorrect usage for both devel and V11 for most cases not just the return you have here. The one remaining issue, and I will not close until that is fixed, is when you have mismatches for equality/non-equality check with class variables or null (e.g. null==0) is invalid, but we do not report a message for that yet.

@steveicarus @martinwhitaker I am having an issue when trying to get the information for class variables. It looks like the class scope does not have the variables so they are not found and test_width ends up just using some defaults that do not impact code generation, but is preventing me from adding the last check to verify that the equality/non-equality operators cannot check that both sides are class variables/null or are normal variables. There are comments in the test br_gh440.v that points to the actual source code where the check is commented out. Do either of you have an idea how to get this information?

steveicarus, 2020-12-28:
> This now correctly reports the incorrect usage for both devel and V11 for most cases not just the return you have here. The one remaining issue, and I will not close until that is fixed, is when you [...]

Would the information you seek not be in the class definition (the class_def() method)?

caryr, 2020-12-28:
Yes, of course that was my thought, but it's not obvious how to get what is needed since this is not a scope so a bunch of extra work is needed. I'm also not certain the definition has the information in the correct format.

caryr, 2020-12-28:
I commented out the check in do_elab_and_eval(), netmisc.cc to resolve the issue in #445 until the class information is found correctly.

martinwhitaker, 2020-12-28:
I think this is another example of what I found when looking at issue #292. If the handling of class properties and normal variables could be unified, a lot of gaps would be filled.

caryr, 2020-12-28:
Okay, it looks like a real mess and I'm not certain I have time or the desire to get all this class stuff working correctly and unfortunately it sounds like we have a bunch of different reports that are related to this so it is important not to mention it is going to limit what @svenka3 can implement.

caryr, 2021-02-17:
This should now be fixed in devel. I updated symbol search to find class properties correctly. There are likely places the code needs to be changed to use this and these may be some of the other related report. I will look at this over the next few days.

closed 2021-02-17 by caryr

svenka3, 2021-02-19:
Thanks for the fix Cray. I confirm that this is indeed fixed. I see an error as below:

```log
iverilog -g2012 ivl_uvm_i440.sv 
ivl_uvm_i440.sv:5: Error: Class/null r-value not allowed in this context.
1 error(s) during elaboration.
```

Perhaps we can improve the error message indicating wrong return value/type?

Thanks
Srini

svenka3, 2021-02-19:
Maybe that "r-value" indeed says that - just have to get used to this terminology :-)

caryr, 2021-02-19:
I can look to see if the message can be improved, but if I remember correctly it's not actually checked where we know this is a return value so the R-value (yes compiler terminology) was the best idea I had at the time.