steveicarus/iverilog #441: SV: assertion error: (darray), function elaborate_expr, file elab_expr.cc, line 2402.

svenka3, 2020-12-27:
I get below error:

```log
iverilog -g2012 ivl_uvm_i441.sv
Assertion failed: (darray), function elaborate_expr, file elab_expr.cc, line 2402.
sh: line 1: 34107 Done                    /usr/local/lib/ivl/ivlpp -L -F"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg2222a612f" -f"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg222a612f" -p"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrli222a612f"
     34108 Abort trap: 6           | /usr/local/lib/ivl/ivl -C"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlh222a612f" -C"/usr/local/lib/ivl/vvp.conf" -- -
```

Code:

```verilog
package t;

  class c;
    virtual function c create (string name=""); 
    return null; 
  endfunction

  function c clone();
    c tmp;
    tmp = this.create();
   return(tmp);
endfunction

endclass : c
endpackage

module m;
  import t::*;

  initial begin
  end
endmodule 

```

Thanks

caryr, 2020-12-28:
This is the line that is failing `tmp = this.create();`

steveicarus, 2021-02-08:
OK, there are two bugs here. Icarus Verilog PECallFunction::elaborate_expr(...) is assuming the argument is a darray, which is not correct, and that is the assertion that we see. Relax that assertion and the deeper problem is that Icarus Verilog seems to be not handling the default argument. Call this.create(...) with an explicit argument, and it works. Rely on the implicit argument and it assertion in a different place.

Still working on it.

caryr, 2021-02-17:
I've tweaked the code a bit to resolve #440 and there could be conflicts so make sure you pull from head.

PR #779 (2022-10-14): Handle default argument values for class function method calls 
For class function method calls currently only as many arguments as have
been supplied are elaborated. Any trailing arguments that might have default
values are skipped. This will trigger an assertion later on in the vvp code
generator backend.

Fix this by making sure that all arguments of the function are evaluated.

Note that this already works correctly for class task method calls.

This resolves #441
closed 2022-10-14 by steveicarus