steveicarus/iverilog #442: SV: final block not executing on zero-delay task

svenka3, 2020-12-27:
Few notes on below code:

1. Syntactically zero-delay task call inside final block just compiles - I was hoping to get a warning
2. Call to t() in final block does NOT seem to execute 

```log
m From initial block
m.t
m                   10
```
I didn't see the m.t display from final() block call.

```verilog
module m;

  task t;
    $display ("%m");
  endtask : t

  initial begin
    $display ("%m From initial block");
    t();
    #10;
    $display ("%m ", $time);
    $finish(1);
  end

  final begin
    t();
    $display ("%m From final block");
  end
endmodule 

```

caryr, 2020-12-27:
Note the real issue her is that you should not be able to call a task from a final block since a final block is defined to only support operations allowed in a function declaration and a function declaration is not allowed to enable tasks. #443 uses the correct functionality so we need to report the task enable as invalid and add support for calling a function. I need to check if no delay tasks are supported by commercial tools to make sure there is not common exception.

caryr, 2020-12-27:
No delay tasks are allowed in a final block by commercial tools

caryr, 2020-12-27:
We are not allowing tasks in final blocks except there was a sneak path for task with no arguments where this was not being checked. For now I am adding the check to this alternate path and not allowing task calls from a final block. If you would like us to match the commercial tools reopen this report.

closed 2020-12-27 by caryr