steveicarus/iverilog #445: SV: default value for function argument as null

svenka3, 2020-12-28:
I get:

```log
ivl_uvm_i445.sv:4: Error: Class/null r-value not allowed in this context.
ivl_uvm_i445.sv:4: error: Unable to evaluate null as a port default expression.
2 error(s) during elaboration.

```

Code:

```verilog
package t_pkg;

  virtual class uvm_component;
    function new (string name = "uvm_component", uvm_component parent = null);
    endfunction : new 
  endclass

endpackage : t_pkg

module m;

  import t_pkg::*;

  initial begin
    $display ("%m From code block");
  end
endmodule 

```

svenka3, 2020-12-28:
I believe this worked till last evening, so has to do with recent fixes I guess. I had to make fixes in 5-6 places in IVL_UVM lib code to workaround this today. Would be great if we can explore a way to add IVL_UVM to your testsuite to avoid future breakages. 

Thanks
Srini

steveicarus, 2020-12-28:
Our test suite (ivtest) is there for all to see. Propose changes, and we'll
consider them.

caryr, 2020-12-28:
Given that code was added last night you would be correct. This is from the fix for #440. This is likely all relate to the current inability to currently find variables in class scopes. I assumed this only applied to the equality/non-equality operators. I can disable this elaboration fix for now and enable once the class lookup is working correctly. I still have a fall back check in the code generator to catch the original issue. It's just I could give much better warnings earlier.

Regarding the test bench, for me it has been a trade off of working on bugs versus TB work plus add in the sv-tests integration I have been working on. Like Steve said you are free to add this yourself. I have even offered some advice along this path and said I would provide support as needed.

caryr, 2020-12-28:
Fixed in devel and 11 by reverting one of the checks in #440

closed 2020-12-28 by caryr