steveicarus/iverilog #449: Verilog - $finish(1/2) does not work as expcted

svenka3, 2020-12-28:
Hi,
  Maybe a known limitation, but wondering why Icarus does not print additional info on

```verilog
$finish (1);
$finish (2);
```
as per LRM Table 20.1. 

Thanks

caryr, 2020-12-28:
I have a fix where both 1 and 2 report the file/line information and the simulation time. The memory and other stuff for 2 is not printed yet. This broke a huge number of tests because of gold or PASSED not being the last thing printed so I am fixing the tests to use $finish(0) where appropriate. I'm not sure how easy it is going to be to get the memory information. I assume the file/line information is the most important with the simulation time a distant second. I'll get to the missing stuff in 2 once things are clean if it is not too much work.

svenka3, 2020-12-28:
Fantastic, do.let me know if I can help in text fixes here.

Thanks

caryr, 2020-12-28:
ivtest has been updated and push along with correct support for 0/1 and at the moment 2 just outputs the same as 1. This was only added to devel since it changes the output of the simulator though it is trivial to switch $finish to $finish(0) to get the same output. Both $stop and $finish should have this new functionality.

svenka3, 2020-12-29:
Inspired by your quick turnaround, I attempted to add support for Sim stats - Memory, CPU Time, User Time (as needed by $finish (2) ). With bit of googling and trials, below is what I get:

```log
Simulation Statistics:  Memory: 3600 (KB) User Time: 0.010242 (s) CPU Time: 0.002378 (s) 
../../ivl_uvm_tests/ivl_uvm_test_1/ivl_uvm_test_class.sv:44: $finish(2) called at 10000 (10ps)
```

I would openly admit am no seasoned SW developer, so code maybe sub-optimal. I have uploaded the modified sys_finish.c file here - https://github.com/svenka3/ivl_uvm/tree/main/tmp/

Kindly review and comment.

Review with Icarus dev team. Few notes:

1. Code largely borrowed from the internet, credits in the copyright header
2. One function is unused - getCurrentRSS - not clear if this is useful
3. Memory is reported as KB, maybe we should do MB?
4. Only tested on MAC, other OS requires build and testing

svenka3, 2020-12-29:
Added Linux64 support to that code (Tested on CentOS7). Windows is still open.

caryr, 2020-12-29:
I had started something already, but you are at the crux of the issue which is getting the information across all the systems Icarus could be used on and it's much more than the ones we currently test in CI. To be done correctly this likely will require configure checks, etc.

caryr, 2020-12-29:
FYI a BSD variant is the big missing CI item and I just checked it is still not supported.

steveicarus, 2020-12-29:
We may eventually more deliberately constrain the systems supported,
but I agree with Cary that this particular feature is really tough to
support, even across the big 3 target systems (windows, Linux, macos)
and BSD doesn't even have a CI instance to support it.

svenka3, 2020-12-30:
I am not sure if I follow this discussion deep enough especially the one on CI, but BSD does seem to have rusage API, 

https://www.freebsd.org/cgi/man.cgi?query=getrusage&sektion=2

I will update my local file with that and will try and add Windows support too. I can't test BSD though as I don't have access to it.

Cheers

caryr, 2020-12-30:
Since @svenka3 is taking over adding the functionality for finish(2) I am removing myself from the assignment list.

The CI discussion is that there is no BSD host in CI so this needs to be manually tested in a BSD VM.

FYI at one point in time I had Icarus compiling on Solaris, but have not tested that or any of the other more corner OS variants in a long time. As a minimum we should expect things to compile and correctly report on Linux, BSD, MAC and windows. You should likely add a message asking the user to report their system if you do not generate the correct information. This will allow use to add support for missing systems as needed.

Also make sure you check that this information is being tallied correctly. For example the LXT2 dumper runs in a separate thread and at some point in time we should switch FST and maybe also VCD to use a separate thread. We do not want to drop this extra thread CPU/memory information. This will likely require some experimenting on your side to verify that you are getting the correct total time and memory. I also agree memory should be reported in MB or are you using MiB (1e6 versus 1024*1024)? We just need to use the correct suffix for the units actually used.