steveicarus/iverilog #450: SV: class method calling from another class

svenka3, 2020-12-28:
Hi,
  I am unable to call a method declared inside a class from another class via handle/object. The same works in a module context. I do recall there was/is an old ticket raised, I thought that was fixed.

```verilog
package t;

  class c;
    virtual function put (i);
    endfunction
  endclass : c

  class test_c;
    c c_0;

    function new ();
      c_0 = new();
      c_0.put(1);
    endfunction : new

  endclass : test_c

endpackage

module m;
  import t::*;

  initial begin
  end
endmodule 

```

Log:

```log
ivl_uvm_i450.sv:13: error: Enable of unknown task ``c_0.put''.
1 error(s) during elaboration.

```

martinwhitaker, 2020-12-28:
Try adding label:bug in the filter box - it makes it easier to find unfixed bugs. This is issue #292. Closing as a duplicate.

closed 2020-12-28 by martinwhitaker

svenka3, 2020-12-28:
Reading the linked issues, looks like the fix is quite involved and may take a while. Is that a fair conclusion from a user perspective?

I need to know this so that I can workaround and make some progress on IVL_UVM. If you intend to fix it in a week or so (yes, appreciate it is a freelance sort of work on development side), I would wait.

Thanks