steveicarus/iverilog #452: SV: Enum var - assigning int

svenka3, 2020-12-29:
Hi,
  For enums - assigning an int value requires explicit static casting in SV LRM. However many tools relax this and issue a warning. In Icarus this throws an error as below, can you consider demoting it into a warning and do implicit conversion?

The second issue is - type cast is NOT yet supported, so all the more reason to consider this request.

```log
ivl_uvm_i452.sv:8: error: This assignment requires an explicit cast.
1 error(s) during elaboration.
```

```verilog
package t;
   typedef enum {IDLE, APB_WR, APB_RD} apb_kind_e;

  class apb_xn;
    rand apb_kind_e kind;

    function new ();
      kind = $urandom ();
    endfunction : new 

    virtual function void print ();
      $display ("%m kind: %0d ", kind);
    endfunction : print 
  endclass : apb_xn

endpackage

module m;
  import t::*;

  initial begin
    $finish (2);
  end
endmodule 

```

caryr, 2020-12-29:
My opinion is casting should be added to support this and we should not reduce the type safety to a warning. After decades of working in this industry I know that most people regard warnings as white noise. Of course the explicit cast is only safe if the value is verified to be in range. Much better to use $cast(), but that is also not implemented in Icarus.

jotego, 2022-10-27:
I am facing the same problem with this little code:

```
    enum int unsigned { SRESET = 0, SIDLE, S0, S2, S4, S6, SRMC_RES} busPhase, next;
    
    always @* begin
        case( busPhase)
        SRESET:     next = SIDLE;
        SRMC_RES:   next = SIDLE;           // Single cycle special state when read phase of RMC reset
        S0:         next = S2;
        S2:         next = S4;
        S4:         next = busEnd ? S6 : S4;
        S6:         next = isRcmReset ? SRMC_RES : (canStart ? S0 : SIDLE);
        SIDLE:      next = canStart ? S0 : SIDLE;
        default:    next = SIDLE;
        endcase
    end 
```

iVerilog reacts in the way described in this 1-year old issue. Has there been any progress about this? Is it possible to make the cast now? Is it possible to demote the error to a warning?

caryr, 2022-10-28:
It looks like your code always assigns an enum value to next, so what is the actual failure?

jotego, 2022-10-28:
The problem is that iVerilog complains about `This assignment requires an explicit cast.` for several of the lines in the case statement. Even though all the lines have the same enum data type on both sides of the assignment. I don't see why a cast is needed there. It might be because the enum is mapped to integers, as this issue suggests.

caryr, 2022-10-28:
Thanks for the clarification.

DvvCz, 2023-11-17:
Having this problem because of my version desync. My project's CI uses `iverilog` from ubuntu which is presumably outdated to the point of enum casting not being available, while my current updated windows `iverilog` forces me to cast the enum..