steveicarus/iverilog #453: SV: enum printing via %s

svenka3, 2020-12-29:
Hi,
  Similar to #452 - to print enum, SV LRM has:

$display ("%s", kind.name());

Now this is NYI (Not Yet Implemented), fair enough.

Some tools allow

$display ("%s", kind);

And prints ASCII equivalent of the integer value underneath. This is very handy for debug. Can you consider this please?

Thanks
Srini

svenka3, 2020-12-29:
label: Enhancement

caryr, 2020-12-29:
Does %d work? Seems like a reasonable request.

caryr, 2020-12-30:
So the complication is the variable created by the enumeration and passe to $display() does not know it is an enumeration. All the access control, etc. is handled by the compiler so there is no way to determine this is an enumeration and should be handled differently. This also likely makes $cast() complicated to implement. It might be easier to implement the name() method to work correctly. The name method is implemented, but does not work with display or when assigning to a string variable and I can find no test that actually uses it so it was likely only partially implemented.

svenka3, 2020-12-30:
> Does %d work? Seems like a reasonable request.

Yes but prints as 0/1/2 - sort of expected, but defeats the purpose from debug perspective

svenka3, 2020-12-30:
> So the complication is the variable created by the enumeration and passe to $display() does not know it is an enumeration. All the access control, etc. is handled by the compiler so there is no way [...]

Using name() is perhaps 100% LRM compliant as well.

Thanks

caryr, 2020-12-30:
So I am confused. How is printing 0/1/2 defeating from the debugging perspective, but displaying the ASCII equivalent of the underlying integer for %s handy for debug? How is this not the exact same information? Yes, name would be better since it gives the actual symbolic name. What are you actually asking for that is different from what you can get with %0d?

svenka3, 2020-12-30:
Maybe an example will help:

```log
kind: IDLE name: IDLE DEC: 0
```

Code:
```verilog
typedef enum {IDLE, APB_WR, APB_RD} apb_kind_e;

module m;
  apb_kind_e kind;

  initial begin
    $display ("kind: %s name: %s DEC: %0d", kind, kind.name(), kind);
  end
endmodule
```

caryr, 2020-12-30:
Okay, kind using %s that displays the label is not possible without knowing that the variable is an enumeration and Icarus does not currently have support for that. Your best chance to get something working is the name() method. With that we can at least pass the variable and the type information.

steveicarus, 2020-12-30:
I think implementing <value>.name() is the right answer, and is entirely doable. I think it is definitely the right answer to this PR.

closed 2020-12-30 by steveicarus