steveicarus/iverilog #460: reg dynamic compile time crash

fogjava, 2020-12-31:
Hi, 
I tried below case :
module test;
reg a[];
reg b[];
initial begin
b = new[3];
end
endmodule

it will crash:
ivl: eval_object.c:34: darray_new: Assertion `ivl_type_packed_dimensions(element_type) == 1' failed.

I looked at the code, in PWire::elaborate_sig . found that for reg a[],  packed_dims will always be empty.

BTW， could you please tell me how to use gdb with iverilog

Thanks

svenka3, 2020-12-31:
I maybe wrong, but I believe Dynamic array is not supported yet

caryr, 2020-12-31:
Dynamic arrays are supported, but the assertion is that the packed dimension is more than a single bit. Not sure why that is. From the comments it looks like the intention was to check that we do not have more than one dimension so zero or one should be supported. I'll take a look.

fogjava, 2020-12-31:
Hi Cary,
Here maybe need handle it as int a[],  for this , I see it will made a packed dimension for it.

caryr, 2020-12-31:
I have a fix and am testing now. Yes, switching to an int will fix the issue since it has a packed dimension.

caryr, 2020-12-31:
Fix pushed to development.

closed 2020-12-31 by caryr

fogjava, 2021-01-06:
> Fix pushed to development.

Hi Caryr,
Where can I find your commit?
Thanks

caryr, 2021-01-06:
It is in the main/development branch on github.