steveicarus/iverilog #461: SV: crash on file with multiple packages, scope resolution issue?

svenka3, 2021-01-01:
Hi,
  I get a crash when a class from one package gets extended in another package (import done). 

```log
sh: line 1: 46637 Done                    /usr/local/lib/ivl/ivlpp -L -F"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg22eb7c10d" -f"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg2eb7c10d" -p"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrli2eb7c10d"
     46638 Segmentation fault: 11  | /usr/local/lib/ivl/ivl -C"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlh2eb7c10d" -C"/usr/local/lib/ivl/vvp.conf" -- -
```

```verilog
package u_lib;

  class uvm_void;
  endclass 
  class uvm_object extends uvm_void;
  endclass 

endpackage

package t1_pkg;
  import u_lib::*;

  class drvr extends uvm_object;
  endclass
endpackage : t1_pkg

module m;
  import t1_pkg::*;

  initial begin
    $finish (2);
  end
endmodule 

```

svenka3, 2021-01-02:
Quick note - the issue is in line:
```verilog
  class drvr extends uvm_object;
```
If I make it as a simple class (without a base class), it works.

Thanks

martinwhitaker, 2021-01-03:
I can't reproduce the fault with the current head of the master branch.

martinwhitaker, 2021-01-03:
But the compiler is terminating early without an error message, so there is something wrong.

svenka3, 2021-01-03:
I tried it after a git pull, compile (of iverlog) on MAC, I get slightly different number, but a crash as below

```log
sh: line 1: 53467 Done                    /usr/local/lib/ivl/ivlpp -L -F"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg2358f568f" -f"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlg358f568f" -p"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrli358f568f"
     53468 Segmentation fault: 11  | /usr/local/lib/ivl/ivl -C"/var/folders/rk/7dbwr18d6f9g7w9yqb06s6yh0000gn/T//ivrlh358f568f" -C"/usr/local/lib/ivl/vvp.conf" -- -
```

martinwhitaker, 2021-01-03:
Fixed in both the master and v11 branches.

closed 2021-01-03 by martinwhitaker