steveicarus/iverilog #464: SV: interface as port item not supported?

svenka3, 2021-01-02:
Hi,
  While SV interface seems to work in Icarus as local elements, passing it via port list is not working. Is this a known limitation? Sounds not too hard to support especially since driving interface signals seems to work (didn't check waves, but log I did).

Below is a test:

```verilog

interface apb_if (input bit pclk);
   logic        psel;
endinterface : apb_if

module m (apb_if if_0);
endmodule : m 

```

Log:

```log
apb_if.sv:6: syntax error
```

martinwhitaker, 2021-01-02:
You've asked this before - see issue #166.

I think there's quite a lot of work required to support this. I'm happy to be proved wrong though - PRs are welcome!

Closing as a duplicate.

svenka3, 2021-01-02:
Martin,
  With due respect, these two are different requests - virtual interface is far ahead in the game I am quite sure. But interface as port list is not the same thing. I would respect your analysis on the effort for sure.

Cheers

martinwhitaker, 2021-01-02:
OK, I stand corrected (although I did answer the question in that issue). I was conflating it with the old feature request on SourceForge. As we are migrating away from SourceForge, let's reopen this.

jotego, 2022-08-19:
I would be interested on this feature too.

UweBonnes, 2023-06-22:
I would be interested on this feature too. I got a project using this feature excessive and trying to replace this feature is very tedious and error probe.

caryr, 2024-01-23:
Here is another example of code already in the tests: ivtest/ivltests/sv_interface.v

ldoolitt, 2024-02-13:
If someone starts working on this, maybe they'd like to see another -- slightly more exotic -- common use of interfaces in ports.  The port is defined not as the whole interface, but as a modport thereof.
So after interface `bound` is defined, the inner module says `module counter(input clk, bound.send send);` and the instantiating module says `counter c(.clk(clk), .send(b.send));`
I'll attach a stylized example that compiles fine on Vivado and Verilator, but sadly not on Icarus or Yosys.  I also filed an [issue](https://github.com/YosysHQ/yosys/discussions/4170) for Yosys, and was told not to hold my breath. If the example's syntax is not standard-conforming SystemVerilog, please let me know.
If anyone is interested, I can turn in a pull-request to put this file in ivtest/ivltests.  Like sv_interface.v, it fails to compile on (today's git master) iverilog.
```
$ iverilog -g2005-sv submodport.sv 
submodport.sv:15: syntax error
submodport.sv:1: Errors in port declarations.
submodport.sv:22: syntax error
submodport.sv:22: Errors in port declarations.
```
[submodport.txt](https://github.com/steveicarus/iverilog/files/14255402/submodport.txt)