steveicarus/iverilog #465: SVL interface - parsing support for clocking block, modport?

svenka3, 2021-01-02:
Hi,
  I understand clocking block and modport are NYI in Icarus. It gives a "sorry" message as well. I am wondering if we can add parsing level support and maybe issue a warning if it is not too much work. Rationale - with some IVL_UVM tests passing in Icarus I am trying to move to some simple, realistic designs. Most of the practical code in SV interface uses clocking block (and modport as distant next if you will). As of now, we have to add `ifndef IVL_UVM to get past this issue. Will be easier to port code to Icarus if this code is parsed and ignored. Yes, we will need to fix our UVM driver etc to not use clocking block, but interface can be as-is.

Thanks

svenka3, 2021-01-04:
I see modport already getting parsed. I took a baby step into adding clocking block parse-only support. As of now I can parse the below code inside interface.

```verilog
  clocking dcb_1 @ (posedge clk);
  endclocking 

  clocking dcb_2 @ (posedge clk);
  endclocking : dcb_2

  clocking dcb_3 @ (posedge clk);
    //output sig1;
  endclocking : dcb_3

```

Need to add clocking block signals support. Is it appropriate to seek help via Discussions forum or sf-devel list?

caryr, 2021-01-04:
See the comment I made to #468 regarding just ignoring language constructs be default. I think that is the wrong thing to do.

svenka3, 2021-01-05:
Would adding a warning and implement the parse-only support be a step forward or would you rather hold this till you (one of you I mean) get bandwidth to do full support?

caryr, 2021-01-07:
Adding an error that you can ignore with a flag is good enough. I doubt we will have time for this anytime soon.