steveicarus/iverilog #472: cannot produce output file on a different directory in windows

Ali-cr8, 2021-01-10:
*Environment variable was set

iverilog is recognized as a  command  in any directory but using command  : 
iverilog -o test test.v doesnt produce a file unless its in the bin file.

gtkwave , vvp commands are working fine regardless of the directory.

caryr, 2021-01-10:
None of the main developers work on Windows so as a minimum we are going to need more information.

What environment variable is set and what is it set to?
Where did you get the executable?
What version of windows are you using?
What are the contents of the bin directory?
Have you tried a different filename (e.g test.vvp)?
What is the Verilog file you are using to test this?
When run in the bin directory does it produce a file that vvp runs successfully and produce the correct results?

All this seems to be some kind of setup issue since iverilog can at last under one case produce the correct output. We will likely need more information and probably help from others before we can resolve this.

caryr, 2021-01-20:
No reply so closing as invalid

closed 2021-01-20 by caryr