steveicarus/iverilog #476: Question regarding 'always' block naming in translated vhdl

svenn71, 2021-01-11:
I have an always block like this

```
always @(posedge clk)
begin : p_block_name
    ... verilog ...
end
```

I would expect this to be translated into

```
p_block_name : process (clk)
begin
    ... vhdl ...
end process;
```
but the 'p_block_name' does not seem to be attached to the process in vhdl. It is still anonymous.

caryr, 2021-01-11:
We have had no one working on the VHDL target for a long time. I have no idea how hard this is to implement.

caryr, 2021-02-09:
I have pushed a fix that may solve your issue. I do not have GDHL installed so cannot test if this breaks anything else, but for the simple Verilog I wrote it did produce code that matched what you expected. Please give the latest development a try and let me know if this works and if there are any issues.

svenn71, 2021-02-09:
The always names are picked up correctly from the verilog files I have and translated to process names in VHDL. Thank you.

caryr, 2021-02-09:
Great. You are welcome.

closed 2021-02-09 by caryr