steveicarus/iverilog #477: Compilation error when slicing packed array with width specifier

logicmonkey, 2021-01-12:
`module m(y, a);
  output logic  [7:0] y;
  input  wire  [39:0] a;

  logic [3:0][9:0] t;

  assign t = a; // 4x10b packed

  assign y = t[2][9 -:8];  // ivl failure

  // assign y = t[2][9:2]; // but this works
endmodule
`
pb@fatcat lm_lib % iverilog -g2005-sv m.sv
m.sv:9: assert: netlist.cc:2508: failed assertion packed.size() == 1
sh: line 1: 22524 Done blah blah

caryr, 2021-01-12:
I can confirm this is failing with the latest development. The constant part select and the up indexed part select are doing the correct thing. It's only the down indexed part select that is failing. It looks like the up indexed part select code was updated, but the down is using out dated code. I'll fix this.

caryr, 2021-01-12:
This is fixed in development and V11.

closed 2021-01-12 by caryr