steveicarus/iverilog #478: Enum member name treated as wire in port binding

elspv, 2021-01-13:
In the module below, iverilog version "12.0 (devel)" on Windows run with -Wall reports warning "implicit definition of wire 'IDLE'." for the second to last line of the module below.
```
module bug(input logic clk, input logic rst);
    enum logic[0:0] {
        IDLE,
        ACTIVE
    } state;
    always @(posedge clk, negedge rst) begin
        if (rst) state <= IDLE;
        else case (state)
            IDLE: state <= ACTIVE;
            ACTIVE: state <= IDLE;
        endcase
    end
    other other(state == IDLE);  // <-- TREATS IDLE AS IMPLICIT WIRE
endmodule
```
Running the above in a test bench for a few cycles shows that iverilog includes a signal named IDLE (fixed at 1'bz) in module bug, and the input to module other is always undefined.  

Inserting line `wire is_idle = state == IDLE` and replacing `state == IDLE` in the port binding with `is_idle` eliminates the warning, the spurious IDLE signal, and undefined input to other.

caryr, 2021-01-13:
There is also an issue when assigning the state variable using a blocking assignment. The compiler thinks it needs a cast.

caryr, 2021-02-14:
The issue was that module port connections are defined to implicitly declare their port expressions as wires if they have not already been defined. This routine was not checking for the enumeration constants so the enum constant was getting implicitly declared as a wire. This is fixed in development.

closed 2021-02-14 by caryr