steveicarus/iverilog #480: VVP Error Message

SteveMillman, 2021-01-15:
Can you please shed light on how to fix this error:

[image]

Thank you,

Steve

martinwhitaker, 2021-01-15:
The compiler - runtime interface is guaranteed to be stable for the lifetime of a major release, but not between major releases and not in the development (master) branch. So, either:
 - use a `vvp` built from the same stable branch or development snapshot  as the `iverilog` used to compile the code (in this case the v11 branch)
 - compile the code with an `iverilog` taken from the same stable branch or development snapshot as the `vvp` being used (in this case an unknown snapshot of the master branch)

caryr, 2021-01-15:
And specifically for your case xor.exe was generated with V11 of the compiler, but you are trying to run this using the development (V12) version of vvp. Like Martin said the version of the compiler and run time must match. FYI Icarus allows a suffix to be specified during compilation so multiple version of Icarus can coexist at the same time and will not interfere. Also, xor.exe should list which vvp it expects to use at the beginning of the file (suffix if used included). If this matches the vvp you are actually using then someone likely updated Icarus between when this was compiled and run.

SteveMillman, 2021-01-15:
Ok, mystery solved! The student's software was all consistent. The problem was that the student got an executable from somebody else and that person was on a different version of the tool.

Thank you so much for helping me out on this!

caryr, 2021-01-16:
You are welcome.

closed 2021-01-16 by caryr